kenjdavidson/react-native-bluetooth-classic #122: Is Bluetooth Device clear() deprecated?

GuillermoGaete, 2021-07-29:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android

**Application Environment**
Provide information about your development environment:
- React Native version: 0.64.2
- RN Bluetooth Classic version: ^1.60.0-rc.16

**Describe the bug**
I am trying to use clear function in order to [docs](https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/#clear) 

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to use clear with connected device.

**Expected behavior**
Clear the buffer message queue. 

**Additional context**
Error captured:

[TypeError: device.clear is not a function. (In 'device.clear()', 'device.clear' is undefined)]

Is a bug?

kenjdavidson, 2021-07-29:
That would indeed be an oversight when wrapping the `BluetoothModule` methods within the `BluetoothDevice` class; it's definitely missing the clear method.

I suggest using the `RNBluetoothClassic.clearFromDevice(address)` method until I get a chance to add it.  

Feel free to submit a pull request with the added method and I can do a release at some point.

GuillermoGaete, 2021-07-29:
Thanks for fast response, i've tried with `RNBluetoothClassic.clearFromDevice(address)` and it works fine!

kenjdavidson, 2021-08-23:
Should be resolved with latest PR.
Will release this evening.