kenjdavidson/react-native-bluetooth-classic #123: getConnectedDevices:rejecter: is not a recognized Objective-C method

tiagotwistag, 2021-08-09:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Ipad 8
- OS: ios 14.2

**Application Environment**
Provide information about your development environment:
- React Native version: 0.64.1
- RN Bluetooth Classic version: 1.60.0-rc.16

**Describe the bug**
When running the project on iOS the following error pops when calling the method **getConnectedDevices**,
`BluetoothScanner[25233:339267] [native] Exception 'getConnectedDevices:rejecter: is not a recognized Objective-C method.' was thrown while invoking getConnectedDevices on target RNBluetoothClassic with params (
    194,
    195
)
callstack: (
	0   CoreFoundation                      0x00007fff20422fba __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007fff20193ff5 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff20422de3 +[NSException raise:format:] + 0
	3   Foundation                          0x00007fff207749b0 -[NSAssertionHandler handleFailureInFunction:file:lineNumber:description:] + 166
	4   BluetoothScanner                    0x0000000108e3c08f -[RCTModuleMethod processMethodSignature] + 1023
	5   BluetoothScanner                    0x0000000108e43ebe -[RCTModuleMethod invokeWithBridge:module:arguments:] + 190
	6   BluetoothScanner                    0x0000000108e49007 _ZN8facebook5reactL11invokeInnerEP9RCTBridgeP13RCTModuleDatajRKN5folly7dynamicEiN12_GLOBAL__N_117SchedulingContextE + 1495
	7   BluetoothScanner                    0x0000000108e48825 _ZZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEiENK3$_0clEv + 165
	8   BluetoothScanner                    0x0000000108e4876c ___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke + 28
	9   libdispatch.dylib                   0x000000010c2d6578 _dispatch_call_block_and_release + 12
	10  libdispatch.dylib                   0x000000010c2d774e _dispatch_client_callout + 8
	11  libdispatch.dylib                   0x000000010c2ddf9a _dispatch_lane_serial_drain + 796
	12  libdispatch.dylib                   0x000000010c2dec67 _dispatch_lane_invoke + 436
	13  libdispatch.dylib                   0x000000010c2eaa7a _dispatch_workloop_worker_thread + 872
	14  libsystem_pthread.dylib             0x00007fff603404c0 _pthread_wqthread + 314
	15  libsystem_pthread.dylib             0x00007fff6033f493 start_wqthread + 15`
After some testing I noticed that only this method seems to be having problems to run iOS(also it works fine on android),
any clue what could be the issue or the fix here?

**Expected behavior**
The method to be available and working iOS

kenjdavidson, 2021-08-09:
`getConnectedDevices` is a new method that was added in the latest version, so I must have missed adding the `rejecter` parameter to the method parameters.

**Implementation**
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/88c3a83d95a8fc2bec8c1d762ad7238b4b31f215/ios/RNBluetoothClassic.swift#L378

```
    @objc
    func getConnectedDevices(
        resolver resolve: RCTPromiseResolveBlock,
        rejecter reject: RCTPromiseRejectBlock
    ) -> Void {
```

rejecter is there.

**ObjC Interface**
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/88c3a83d95a8fc2bec8c1d762ad7238b4b31f215/ios/RNBluetoothClassic.m#L97

```
RCT_EXTERN_METHOD(getConnectedDevices: (RCTPromiseResolveBlock)resolve
                  rejecter: (RCTPromiseRejectBlock)reject)
```

rejecter is there.

It looks like it's calling the method with just the rejecter and not the resolver.  This is probably because of the FIRST parameter thing in swift/objective c (which I can't stand).  It probably needs to be updated to `_` so that it picks up the right one.

tiagotwistag, 2021-08-09:
Hey, thank you so much for the fast awnser
I edit the RNBluetoothClassic.swift directly on the node modules and tried to run the project again and seems that the problem persists, I'm not sure if I'm missing something
`
    func getConnectedDevices(
        _ resolve: RCTPromiseResolveBlock,
        rejecter reject: RCTPromiseRejectBlock
    ) -> Void {
`

kenjdavidson, 2021-08-09:
I'm not sure - I'm definitely not an IOS developer and I don't particularly like the Objective C or Swift languages.  I also don't have time at this point to fire it up and play around to try and get it working (I apologize).

If you can sort it out I'll definitely accept a pull request.

Otherwise you'll have to wait until I get time to play around with it some more.

commit cf910ac8 (2021-08-23): Merge pull request #128 from amitferman/master

Fix #123 & Expo Installation
closed 2021-08-23 by kenjdavidson