kenjdavidson/react-native-bluetooth-classic #124: Expo Config Plugin

amitferman, 2021-08-11:
**Mobile Device Environment**
iOS

**Is your feature request related to a problem? Please describe.**
You can't develop on Expo out-of-the-box with react-native-bluetooth-classic since it doesn't have a [Config Plugin](https://docs.expo.dev/guides/config-plugins/) to handle native configuration. 

**Describe the solution you'd like**
A Config Plugin would be a very useful addition to the library. It would also expand its use cases to Expo projects.

kenjdavidson, 2021-08-11:
I can see your points, although I've never used Expo.  The few times I have (probably my own issue) I've messed things up and had to just export the project anyhow.

It looks like there are a bunch of `best practice` expo dependencies that I don't particularly want in the library, to me it looks like creating a `expo-bluetooth-classic` project and maintaining it separately would be best?  Keeping all the extra Expo related dependencies separate.

I'll have to do a little reading, and I definitely don't have time to look into this now.

I can't think of anything specific to this library that would need to be in there, so effectively any `expo-bluetooth` plugin that updated the correct PLIST items would work.  You've googled around and none exist?

Can you think of any other configurations in this library that would need to be included in the plugin?  Since you've obviously got way more experience than I, any thoughts would be appreciated.

amitferman, 2021-08-11:
Thanks for the quick response. The issue (or interesting part) with config plugins is that they were released by Expo about 3 months ago, so most React Native libraries require you to "eject" from Expo and move to pure React Native development. Furthermore, config plugins are essential for using Expo in a project with custom native code, such as yours. This means that for people like me, who want to develop iOS apps on a Windows machine and therefore can't eject from Expo, config plugins might be important.

Though I have only recently started to use Expo and develop on iOS, I might try to create the plugin according to Expo docs since it doesn't seem too complicated and the Expo docs provide an example I could use. To confirm, these:
`<key>UISupportedExternalAccessoryProtocols</key>
  <array>
    <string>com.apple.m1</string>
  </array>`
are the only pList items required for native configuration, correct?

I will let you know if I get it working and if I do, the plugin might be a useful thing to link in your README.

Thanks again

amitferman, 2021-08-11:
That's a great idea. I'll try that as well. Thanks for helping me out with this!

kenjdavidson, 2021-08-11:
No problem, whatever gets more people working faster is cool with me!  

I just don't have the time at this point to really take a deep dive into it - it looks like this won't ever work with the Expo Go app - based on comments in that one library README.   

I'm not sure how I feel about things like Expo in the grand scheme of things, I'd have to take a look back at it now that that they've made it possible to use more functionality (specifically for IOS).  So thanks for opening this!

amitferman, 2021-08-11:
Though it won’t work on the Expo Go app, Expo offers an “Expo Development Service” that lets you install a precompiled runtime with your custom native packages, such as this one. That’s what I’m trying to do and I’ve had some success with it. I think it’s pretty cool that Expo let’s you create your own versions of the Expo Go app by just installing the expo-dec-client non package and building on the cloud, though it’s very new. According to my research, developing iOS on Windows is a rapidly expanding field of services and Expo tries to be an all-in-one solution to this.

Check out this Expo-sponsored video if your interested: https://m.youtube.com/watch?v=Iw8FAUftJFU It explains the functionality very well.

amitferman, 2021-08-19:
After looking back on this problem, I was able to find a solution and opened a pull request accordingly.

closed 2021-08-19 by amitferman

bfarrgaynor, 2025-02-07:
@amitferman sorry did you say you were able to get 'this' (react-native-bluetooth-classic) working using an expo development client? I need to support MFi and Serial bluetooth and I really don't want to eject if I can help it.

kenjdavidson, 2025-02-07:
He added an expo section to the documentation page.  So I'm assuming that it's been working since then.