kenjdavidson/react-native-bluetooth-classic #131: I can't describe this bug, but don't works

IgorDePaula, 2021-09-06:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: moto e6
- OS: [e.g. API19, Nouget, iOS8.1]

**Application Environment**
Provide information about your development environment:
- React Native version: RN 0.64
- RN Bluetooth Classic version: last
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Init a new RN project
2. copy and past the code from docs about scanDevices
3. see the error
Happens only this.

**Expected behavior**
I expected show a list decices available

**Screenshots**
If applicable, add screenshots to help explain your problem.

kenjdavidson, 2021-09-07:
Going to need substantially more information than you've provided:

- link to your repository?
- sample code for the issue?

Based on your environment, the issue could be anything:

- Android 8 (which seems unlikely since I'm sure it's been used here)
- React Native 0.64 (maybe, I'd have to check if they made any breaking changes)

That error message looks like more of a React Native issue, and not a Blueooth Classic library issue - it specifically says you're rendering a child component incorrectly.

IgorDePaula, 2021-09-09:
This is the mistery, I only put script to scan bluetooth and log him, not write components on screen. But I resolved with another library. Sorry, and Thank you very much for your attention.

kenjdavidson, 2021-09-09:
Glad you got it working man, sorry I couldn't help.

closed 2021-09-09 by kenjdavidson

IgorDePaula, 2021-09-09:
No problem. Thank you very much for your attention.

IgorDePaula, 2021-09-10:
I decide try again this lib, but I not got success.

Now is another error:
[image]

The code:

[image]
 Is only this of code. I try connect on ESP32, and sometimes I got error of  wrong pin/password, but Android I try pair or connect and I receive this same error, and not appear prompt to I write some pin.

kenjdavidson, 2021-09-10:
I'll be completely honest, I don't really test or use the `pairDevice` in my app, I trust that the device is already paired through the Android Bluetooth settings.  Have skipping the pairing to see if that's the issue? If it is, I'd need to play around with it a little to see what's up - although not really a top priority for me at this time. 

If you find that the pairing has issues and you fix them, feel free to submit a PR for it.

Anyhow, following the log and the code:
- `to aki` is logged so that's good
- `ESP32` is logged, which I don't see there so I'm assuming it's from the `setDevice` state function
- the `ERROR: IOException` is logged which I'm guessing is coming from the `device2.connect()` catch

You're connecting with the default configuration:
- RFCOMM
- Secure
- ASCII

Looking at the documentation for this chip 
https://www.espressif.com/sites/default/files/documentation/esp32_bluetooth_architecture_en.pdf

it doesn't look like it supports RFCOMM protocol (even thought it's mentioned):
Below are the Classic BT profiles and protocols currently supported in the Host Stack.
• Profiles: GAP, A2DP(SNK), AVRCP(CT)
• Protocols: L2CAP, SDP, AVDTP, AVCTP 

If this is actually the case, there are two options:
- The current `RfcommConnectorThreadImpl` can be copied and modified for `L2CapConnectorThreadImpl` which may work
- The current `RfcommConnectorThreadImpl` can be modified to accept another property to determine `rfcomm` or `l2cap` in which case it would need to be renamed to something better `SerialConnectorThreadImpl`? `ProtocolConnectorThreadImpl`? 

Or something else entirely is going on.

I would suggest you take a look at the documentation and either:
- Update your local install of the library for testing (then update back or submit a PR)
- Use the `Override Auto Loading` feature to create your own `L2CapConnectorThreadImpl` and use it within your own app

anyhow, look forward to more of your troubleshooting.

IgorDePaula, 2021-09-14:
I comprehend, and I would like share a feedback. I tried using the serial bluetooth code on ESP32, and the same code on RN, and I got this error:

[image]

But, any form, I got success using the react-native-ble-manager lib. Thank you for your complete attention.

kenjdavidson, 2021-09-14:
Well ble and classic are two completely different things. That error leads me to believe your Bluetooth adapter was off completely, and you weren't catching the exception.

Glad it's working though.

[Captura de Tela 2021-09-11 às 20 54 54]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F2833993%2F133345262-bb2e44c8-c5d0-4852-b364-1c4609b2f38d.png&data=04%7C01%7C%7Cc21c456a87e245e4270f08d977d47b8a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637672576752297139%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=YzABwEZLEvTBFLL6y34J5Z4bFEhw3d7Z%2FpmRHpac%2FYk%3D&reserved=0>