kenjdavidson/react-native-bluetooth-classic #138: [Question] How to pair, write, read after discovery?

faker007, 2021-10-31:
**Mobile Device Environment**
On which device(s) would you like this feature to be implemented.   If Android only, the IOS implementations must be created as well - keep not of this for your pull requests.
- Device: LG-7
- OS: API 29, Android 10

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
**[Question] How to pair, write, read after discovery?**

**TL;DR:** So, I would like to ask you my brothers, how to pair, read, write with HC-06... I need a working code snippet I really ask you guys humble

Hello, My brothers, I'm currently doing a project about connecting Arduino(HC-06) and RN with this library.

At first, I was so happy, because the other libraries(react-native-ble-manager, react-native-ble-plx) doesnt work well with HC-06, because I guess those libraries are not support in serial communication. (I'm not sure)

But, with this libray, I could discovery but I realise that I don't know know to pair, write, and read...

Perhaps, I can use pre-made connector for on and off to the LED... 

However, I need some extra feature about Firebase...

So, I would like to ask you my brothers, how to pair, read, write with HC-06...

Thanks for reading...

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

faker007, 2021-10-31:
Thanks for reading... I look though docs but I didn't find all in one

**What I tried** is...
`1. const unpaired = await RNBluetoothClassic.stratDiscovery() // it's okay and I got what I want `
`2. const result = await RNBluetoothClassic.pairDevice(element.address); // from one of the unpair variables`

from 2. -> I couldnt pair even, I tried to `RNBluetoothClassic.pairDevice(...)`

faker007, 2021-10-31:
From now, I successfully paired HC-06 with my Android phone. by using this function `RNBluetoothcLASSIC.connectToDevice(...)`

So, I have to figure out how to write and read the data...

kenjdavidson, 2021-10-31:
I would suggest looking at the Bluetooth sample app repository. It's got a fully working example. There are also numerous forks/repos that are using this library you can attempt to follow.

Youd also need to confirm/validate your microcontroller code, which is beyond the scope of this issues list.

closed 2021-10-31 by kenjdavidson