kenjdavidson/react-native-bluetooth-classic #139: Removal of readSubscription problem

Wailimy, 2021-11-10:
Let's say this was done during connect:

this.readSubscription = this.device.onDataReceived(data => this.onReceivedData(data) 

However, this.readSubscription cannot be removed later on when connection is suddenly broken, because it depends on there being a connection. However, when this is not removed, and connection is remake, this results in duplication of data.

kenjdavidson, 2021-11-10:
Ya I can see that.  I guess the options are

- don't throw an exception if attempting to remove listeners and no connection?
- keep native and just catch the exception. To allow app to handle?
- remove subscription before native and let the 3xception throw as it is.

What are your thoughts?
Im leaning the first, for read listeners. Keeping the rest of the listeners the same.

kenjdavidson, 2021-11-25:
v1.60.1-rc18.

closed 2021-11-25 by kenjdavidson

Jose-Gallegos, 2022-01-12:
I have same problem, after connection broken, I attemp to remove the read subscription, but when I reconnect and make a new read subscription I get duplicated read events. It seems the first read subscription was not removed. I'm using v1.60.0-rc20. How can I avoid duplicated events?

kenjdavidson, 2022-01-12:
Need more info on how you're disconnecting. When you get a disconnect event you should kill the subscription? If we kill the subscription on the native side, you'd still have it open on the react side.

What's your full use case for this?

Jose-Gallegos, 2022-01-12:
I am connecting to a RFID Reader (TruTest XRS2). When I shutdown the reader, I get an onDeviceDisconnected event. Inside that event I call remove() for the read subscription for that device. Then I have tried two scenarios, calling disconnect() for that device, and not calling disconnect() for that device (getting same behavior). Then I power on the rfid reader, connect to the device again, subscribe to OnDataReceived, and when I make a reading, I get duplicated onDataReceived event.
I have also tried to not subscribe again to read event, but then I get no event at all.

kenjdavidson, 2022-01-12:
What exactly happens when you attempt to remove the read listener? Previously it would have thrown an exception saying the device isn't connected. But that exception was removed and it should just remove it from he native side regardless of there being a connection.

Are you staying you're still getting the exception?

Jose-Gallegos, 2022-01-12:
Nothing happens, it happen silently, not getting an exception.

kenjdavidson, 2022-01-12:
What is happening silently?

- If your device disconnects you should get a DEVICE_DISCONNECT event
- If you attempt to remove a listener, it will either throw an error or work

You're going to need to debug it and let me know what exactly is happening in both cases, and what you think the appropriate flow is.

In my opinion when you:
- disconnect or get a disconnect
- you should remove the subscription

If you're doing that, what happens?
- Does the native side get the request in the `removeListener` method?

The other option is that if there is a native disconnect we remove the subscription, but in order to do that on the React side, we'd need to create another event that needs to be listened for all the time (internally) and we then need to manage all the subscriptions (internally) which just seems like a hassle and takes options away from the developer.

Anyhow, if you can debug it and let me know what's going on in your case, I'd be happy to accept a pull request.

Jose-Gallegos, 2022-01-31:
Hi, I'm back. I was debugging the native code of the library. First I'll answer your questions:

Q: What is happening silently?
A: The remove() method is silent, not throw an error, I don't know if it is working.

Q: If your device disconnects you should get a DEVICE_DISCONNECT event
A: Yes, I get that event.

Q: If you attempt to remove a listener, it will either throw an error or work
A: Not throw an error, I don't know if it is working.

Q:

> What is happening silently?
> * If your device disconnects you should get a DEVICE_DISCONNECT event
> * If you attempt to remove a listener, it will either throw an error or work

The remove() method is silent, not throw an error, I don't know if it is working. I get that event, then I call remove(), but when I reconnect I get two on read event call. If I shutdown the bluetooth device, power on a reconnect again, I get three on read events. I always call remove() on DEVICE_DISCONNECT event, no error thrown.

> You're going to need to debug it and let me know what exactly is happening in both cases, and what you think the appropriate flow is.> 

Yes, I debug it, I see that when data is received, the emit() method is called one, but I get two on read events.
It seems like the remove() method is not working, but no error is thrown.

Thanks,

José

kenjdavidson, 2022-01-31:
I appreciate the update, but you'll need to debug it a bit more to provide specific details on:

1. What happens on the Native side when you call `remove()`?  
Is this finding the `listener` and then removing it from the listener map?

The `removeListener` method should be getting called with `remove()`
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/30962d819086391e1ccf1ddb263ca587eb516bdf/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java#L1122

which should then call `clearOnDataReceived()` does that happen?

Does the listener count get decreased?

2. What's happening on the React Native (js) side when you call `remove()`?
When you call remove on the subscription it should be calling these two things:
- `_nativeModule.removeListener` which calls to Java
- `subscription.remove()` which calls remove on the _eventEmitter subscription

Are these happening?  If the issue is with `_eventEmitter.remove()` then this is an issue with React Native, and you can attempt to update the version and see how that goes.

3. What happens on the Native side when you receive data?  Are there now two READ listeners in the listener map?