kenjdavidson/react-native-bluetooth-classic #140: Is there a way for us not to use setInterval() to read the data? Continuously reading from device without setInterval()

sunny812546, 2021-11-24:
**Similar To This**

BluetoothSerial.withDelimiter('\r').then(() => {
      Promise.all([
        BluetoothSerial.isEnabled(),
        BluetoothSerial.list(),
      ]).then(values => {
        const [isEnabled, devices] = values;
        ...
      });
      BluetoothSerial.on('read', data => {
        console.log(`DATA FROM BLUETOOTH: ${data.data}`);
     });
});

kenjdavidson, 2021-11-24:
I don't understand the question. Is there a problem with the on read listener? It's been working for me and others that I'm aware of.

If you can't get it working I suggest reading the docs and looking at the sample app.

Or are you saying there's a bug with it?

If there is a bug you need to provide way more details on what and how to replicate.

sunny812546, 2021-11-24:
First of all, I am impressed with this library. Great work, great effort.. 🔥

I just want to know how to read continuous read from a Bluetooth classic device using this library.

Because I used setInterval() for continuous Reading.

Is there any better way to continuously read data from the device via Bluetooth?

kenjdavidson, 2021-11-24:
Yes, it's in the doc:

https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/#ondatareceived

And the sample application:

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/1cba89aaa1181998dbcab7a9549a88a9bb5071a1/BluetoothClassicExample/src/connection/ConnectionScreen.js#L140

kenjdavidson, 2021-11-24:
Also, thank you.

closed 2021-11-24 by kenjdavidson

sunny812546, 2021-11-24:
Now, I am getting continuous read and it is very fast. Thanks a lot.❤❤❤