kenjdavidson/react-native-bluetooth-classic #159: Extend scanning time to allow for device names to be found correctly.

scope2229, 2021-12-13:
**Mobile Device Environment**

- Device: [ Android, ]
- OS: [ API28, Version 10 + ]

**Is your feature request related to a problem? Please describe.**

When scanning for devices on Android the name is not always returned. This is the same for BLE devices and Classic devices. Using `react-native-ble-plx` I can scan for X amount of seconds using `setTimeout` and then stop the scan. This means I can filter out devices that don't contain a name. and only add the devices with a name. With this library I have to prevent devices with only an ID from being added to a list. This means I have to scan multiple times to find devices with the name and that's if they show up. Another alternative is to extend the scan time to 60 seconds, 15 seconds is far to short.

**Describe the solution you'd like**

`startDiscovery()` method should never stop unless/until  `cancelDiscovery` has been called.  A note in the documentation should be made to notify users that trying to scan or connect to a device while scanning may cause an error and `cancelDiscovery()` should be called by the user. 

**Describe alternatives you've considered**

Scanning over and over again until the device is found with a human readable name.

kenjdavidson, 2021-12-14:
That's not how the android adapter works.  You call start discovery and you're at the mercy of how long it lasts, approx 12 seconds.

https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#startDiscovery()

I don't think it's wise to over ride this logic on the js or Android side. It should be used as is within the calling application, in my opinion.

kenjdavidson, 2021-12-14:
With regards to the documentation, it should say that it calls through to the Bluetooth adapter, if it does not we can easily add the link.

closed 2021-12-14 by kenjdavidson