kenjdavidson/react-native-bluetooth-classic #160: How to discover device one by one?

bexoss, 2021-12-29:
This repository's `startDiscovery()` function looks like waiting until discovering finished. 
I want to add device to state one by one when it discovered. 
Is this possible?

Below is react-native-ble-plx example.

```
            manager.startDeviceScan(null, null, async (error, device) => {
                  // error handling
                  if (error) {
                    console.log(error);
                    return
                  }
                  // found a bluetooth device
                  if (device) {
                    console.log(`${device.name} (${device.id})}`);
                    const newScannedDevices = scannedDevices;
                    newScannedDevices[device.id] = device;
                    await setDeviceCount(Object.keys(newScannedDevices).length);
                    await setScannedDevices(scannedDevices);
                  }
  
```

kenjdavidson, 2021-12-29:
You should be able to listen for the **EventType.DEVICE_DISCOVERED** event.  It looks like it's in the Android Native code, but I'm unsure if it's connected to the React JS side of things.

```
@Override
public void onDeviceDiscovered(NativeDevice device) {
      // This wasn't previously an event, but now we can send out and request them
      Log.d(TAG, String.format("Discovered device %s", device.getAddress()));
      sendEvent(EventType.DEVICE_DISCOVERED, device.map());
}
```

I suggest you take a look at the code and play around with it.  If it's not working, or you think there is a better way in which it should work, I'm open to suggestions.  Right now there are three things:

Events:

DEVICE_DISCOVERED - this only happens once per device found.  There was another issue where someone mentioned that Android could find a device multiple times (once without the name and once with) which I didn't account for.

Promise response - this occurs when the `startDiscovery` has completed (or is cancelled) after the appropriate Timeout.  Android docs say it's 20s.

You should be able to listen for DEVICE_DISCOVERED and then cancel (and ignore) the promise when you get the device you want.  But remember, at this point each device is only sent once, so if the comment above is true, this will not be sufficient at times.

Feel free to submit a pull request with any changes you think important.

bexoss, 2021-12-29:
@kenjdavidson Thanks for information. I red your answer and tried to use `onDeviceDiscovered` handler. 

Just simply did

```

const RNBC = RNBluetoothClassic
useEffect(() => {
     RNBC.onDeviceDiscovered((device) => {
      console.log('device discovered.')
      console.log(device)
      // (I should find a way to cleanup listener)
    })
}, [])

```
And It's done! It already implemented like you said.

Now I can see discovered devices quickly. Some of them are duplicated, but I think I can distinguish them. 
Thanks for your kindness, and the effort that managing this library. I cannot find supporting classic bluetooth module like hc-06 except this lib! 

Happy New Year!

kenjdavidson, 2021-12-29:
Glad to hear, stay safe!

closed 2021-12-29 by kenjdavidson