kenjdavidson/react-native-bluetooth-classic #161: Unexpectedly found nil while unwrapping an Optional value

faizanbutt, 2022-01-05:
Hi, I got a project from someone else. It is running fine on Android but giving this error on iOS simulator "react_native_bluetooth_classic/RNBluetoothClassic.swift:60: Fatal error: Unexpectedly found nil while unwrapping an Optional value". The simulator screen blanks out and this is the message on Xcode:
[image]

kenjdavidson, 2022-01-05:
You need to provide the `UISupportedExternalAccessoryProtocols`.  There's a big point about this on the documentation site:

https://kenjdavidson.com/react-native-bluetooth-classic/ios/

I haven't bothered handling this gracefully since if you don't have it, nothing will work.

kenjdavidson, 2022-01-12:
I'm assuming you sorted this out.

closed 2022-01-12 by kenjdavidson

faizanbutt, 2022-01-12:
Yes. Thank you