kenjdavidson/react-native-bluetooth-classic #163: Can multiple devices connect to a single accepting device at the same time?

leecaputoit, 2022-01-17:
Hi, I have been able to successfully perform read/write between two connected devices, but whenever I tried to connect another device to the accepting device, the connection resolves to success yet `getConnectedDevices` only return one entry.

After looking through past issues, it seems that support for multiple connections was indeed added, but I am not sure how to go about enabling it. Any pointers would be appreciated.

leecaputoit, 2022-01-18:
@kenjdavidson Thanks for the quick response and the informative comment! I did not realize I had to call `cancelAccept` and then start accepting again for a thir d device to connect, now it works perfectly. :)

closed 2022-01-18 by leecaputoit

kenjdavidson, 2022-01-18:
Ya. I had planned to allow multiple connections in 9me accept session but never got it done. One at a time for the time being.

It should actually resolve the promise and cancel. Accept after the first one anyhow.