kenjdavidson/react-native-bluetooth-classic #164: onWrite complete event?

PietroGranati, 2022-01-18:
Hi Ken, I'm working with you library, finally get to work with iOS device, my project in a certain point I write an "ESC/POS" file to a thermal printer and they have limited buffer to 1024, before react-native I was working on the same project on Appcelerator and I cannot know when the "write to device is over".
With this library everything is working fine and I got this log:
`2022-01-18 15:38:32.742942+0100 registropresenzevisitatori[33626:2634685] (BluetoothDevice:writeToDevice) Writing @a(L>,0p01&¤gistropresenzevisitatori[33626:2634685] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.743307+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.743355+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.743409+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.743501+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.743545+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.743595+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.743646+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.743783+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.743955+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.744196+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.744466+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.744703+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.745060+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.745203+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.745468+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.745771+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.745940+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.746109+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.746233+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.746517+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.746622+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.746790+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.746948+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.747147+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.747356+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.747553+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.747820+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.748062+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.748233+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.748419+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.748602+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.748746+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.749060+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.749276+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.749404+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.749570+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.749721+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.749859+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.750386+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.750548+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.750927+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.751132+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.751275+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.751410+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 1024 bytes to the device
2022-01-18 15:38:32.751582+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 1024 bytes to the device
2022-01-18 15:38:32.751743+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.751980+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Attempting to send 859 bytes to the device
2022-01-18 15:38:32.752125+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) Sent 859 bytes to the device
2022-01-18 15:38:32.752261+0100 registropresenzevisitatori[33626:2634303] Stream <EAOutputStream: 0x282ed61c0> has space available
2022-01-18 15:38:32.752393+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:writeData) No buffer data scheduled for deliver
2022-01-18 15:38:33.513782+0100 registropresenzevisitatori[33626:2634303] Stream <EAInputStream: 0x10d39e9c0> has bytes available
2022-01-18 15:38:33.514385+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:readFromDevice) Reading device 7005358A until delimiter 

2022-01-18 15:38:35.986659+0100 registropresenzevisitatori[33626:2634303] Stream <EAInputStream: 0x10d39e9c0> has bytes available
2022-01-18 15:38:35.987281+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:readFromDevice) Reading device 7005358A until delimiter 

2022-01-18 15:38:36.428350+0100 registropresenzevisitatori[33626:2634303] Stream <EAInputStream: 0x10d39e9c0> has bytes available
2022-01-18 15:38:36.428947+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:readFromDevice) Reading device 7005358A until delimiter 

2022-01-18 15:38:36.686960+0100 registropresenzevisitatori[33626:2634303] Stream <EAInputStream: 0x10d39e9c0> has bytes available
[... cut ...]
2022-01-18 15:38:37.271630+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:readFromDevice) Reading device 7005358A until delimiter 

2022-01-18 15:38:37.546877+0100 registropresenzevisitatori[33626:2634303] Stream <EAInputStream: 0x10d39e9c0> has bytes available
2022-01-18 15:38:37.547308+0100 registropresenzevisitatori[33626:2634303] (BluetoothDevice:readFromDevice) Reading device 7005358A until delimiter `

Do you think is possible to have a callback or some events to listen to the end of the write?

kenjdavidson, 2022-01-18:
Anything is possible, but it depends on what method of doing this would benefit the most people in the most specific way, it would also need to be completed on Android as well.

In most cases the write buffer, is just a buffer, it has no way to know how many write requests you've sent.  For example, what if you did:

- Send 4000 bytes
- Send 4000 bytes

So we would need to keep track of the number of bytes for each write and while writing accumulate them.  The problem being after:

- 1024
- 1024
- 1024
- 1024 (this would be parts of write 1 and write 2)

In your case, you may not send new write requests until the previous is done, but not everyone would do that, they're probably sending delimited strings which will eventually get sent and not need to worry about it.

To do what you want (properly) we would need to:

- Write a custom Connection that handled sending only X bytes, notifying, then continuing.
- Get the IOS version of the custom Package working

If you feel like taking a stab at the second part (the ability to override the auto loading feature) I would certainly accept a pull request.

kenjdavidson, 2022-01-18:
Or we would need to create a queue of output buffers and have the the write method automatically switch them when the primary output buffer becomes empty.

Write 1
- outBuffer is empty so add there
- write

Write 2
- outBuffer is not empty
- add to queue
- write not required as processing should still be handling outBuffer

onSpaceAvailable
- write out buffer
- If nothing left in out buffer move queue into outbuffer
- Send write_finished event
- allow processing to continue as normal

PietroGranati, 2022-01-18:
I was thinking about it and in the previous implementation I made x 1024 packets and send them, in this way I can keep track of how many but need a complete again.
I'll think about it, thank you for the answers!

kenjdavidson, 2022-01-18:
Right on. I have zero time right now. So it would need to be you looking at it and submitting a pr.

Keep me posted.

PietroGranati, 2022-01-18:
Abolutely! Thank you!