kenjdavidson/react-native-bluetooth-classic #165: Sometime i get double response messages

hungdev, 2022-01-19:
I really really want to say thank you to you. I love your lib. It saves me in my project.
But Sometimes I get double response messages. Could you take a look at my useBluetooth.js file and tell me where is my problem. I think I missed some clear subscription there.
Main File

```
import React, { useState, useRef, useEffect, useCallback } from 'react';
import { View, Text, ScrollView, TouchableOpacity, Touchable } from 'react-native';
import styles from './styles/Home.Styles';
import useBluetooth from 'app/hooks/useBluetooth';
import RNBluetoothClassic from 'react-native-bluetooth-classic';

export default function Home() {

  const onReceivedDataScan = useCallback((data) => {
    console.tron.log('data?.data', data?.data); // ========> Get response here
    alert(data?.data);
  }, []);

  const { connection, setDevice, data, } = useBluetooth(onReceivedDataScan);

  useEffect(() => {
    getBondedDevices();
  }, []);

  const getBondedDevices = async () => {
    console.tron.log('-- GETTING BONDED DEVICES --');
    try {
      const listBondedDevices = await RNBluetoothClassic.getBondedDevices();
      const dv = listBondedDevices?.find(e => e.id === "F4:5E:AB:DB:1F:B9");
      setDevice(dv);
    } catch (err) {
      console.tron.log(err);
    }
  };

  return (
    <View style={styles.container}>
      
    </View>
  );
}
```

useBluethooth.js

```
import React, { useEffect, useState } from 'react';
import { View, Text } from 'react-native';
import RNBluetoothClassic from 'react-native-bluetooth-classic';

export default function BluetoothClassic(onReceivedDataScan) {
  const [device, setDevice] = useState();
  const [data, setData] = useState([]);
  const [connection, setConnection] = useState(false);

  useEffect(() => {
    device && setTimeout(() => connect(), 0);
  }, [device]);

  const addData = (message) => setData(prev => [message, ...prev]);

  const connect = async () => {
    try {
      let connection = await device.isConnected();
      if (!connection) {
        addData({
          data: `Attempting connection to ${device.address}`,
          timestamp: new Date(),
          type: 'error',
        });

        connection = await device.connect();

        addData({
          data: 'Connection successful',
          timestamp: new Date(),
          type: 'info',
        });
      } else {
        addData({
          data: `Connected to ${device.address}`,
          timestamp: new Date(),
          type: 'error',
        });
      }

      setConnection(connection);
      initializeRead();
    } catch (error) {
      addData({
        data: `Connection failed: ${error.message}`,
        timestamp: new Date(),
        type: 'error',
      });
    }
  };

  const disconnect = async (disconnected) => {
    console.tron.log('disconnected', disconnected);
    connect();
    try {
      if (!disconnected) {
        disconnected = await device.disconnect();
      }

      addData({
        data: 'Disconnected',
        timestamp: new Date(),
        type: 'info',
      });

      setConnection(!disconnected);
    } catch (error) {
      addData({
        data: `Disconnect failed: ${error.message}`,
        timestamp: new Date(),
        type: 'error',
      });
    }

    // Clear the reads, so that they don't get duplicated
    uninitializeRead();
  };

  const initializeRead = () => {
    disconnectSubscription = RNBluetoothClassic.onDeviceDisconnected(() => disconnect(true));

    readSubscription = device.onDataReceived(data => onReceivedData(data));
  };

  /**
 * Clear the reading functionality.
 */
  const uninitializeRead = () => {
    if (readInterval) {
      clearInterval(readInterval);
    }
    if (readSubscription) {
      readSubscription.remove();
    }
  };

  const onReceivedData = (event) => {
    event.timestamp = new Date();
    const dataReceived = {
      ...event,
      timestamp: new Date(),
      type: 'receive',
    };
    addData(dataReceived);
    onReceivedDataScan?.(dataReceived);
  };

  return { connect, disconnect, connection, setDevice, data, uninitializeRead };
}
```

Thank you!

kenjdavidson, 2022-01-19:
There is an issue with the HOT reloading of React Native where there are no (that I'm aware of) events that can handle disconnecting and re-connecting when it happens.   So when you're doing active development you need to ensure you manually:
- disconnect/remove
- reconnect

This is a known issue, and something that I haven't looked into in a while.  Generally when an app closes the context is lost, the device is disconnected and subsequently the subscription removed.

When hot reloading occurs, this does not happen, the native side does not get reset and therefore doesn't lose is listeners, while the react side does get reloaded and loses it's reference to the subscription.

I don't know if React Native added events for managing hot reloading like this.

kenjdavidson, 2022-01-19:
If you go through some of the closed issues, you can see some of their comments.

Your code looks ok, nothing really jumps out with regards to removing the subscription.  So this is really the only thing that I can think of - the hot reload.

kenjdavidson, 2022-01-19:
https://reactnative.dev/blog/2016/03/24/introducing-hot-reloading#hmr-api

Give this a shot, you may be able to listen for HOT reload and disconnect/remove when required.

```
module.hot.accept(() => {
  console.log('time changed');
});
```

In your case, you probably want to do this within `useBluetooth` in order to clear the state and make sure it's back to the beginning.

hungdev, 2022-01-20:
Thank you for your support <3. I will close this issue to avoid distracting issues and discussions.

closed 2022-01-20 by hungdev