kenjdavidson/react-native-bluetooth-classic #167: RNBluetoothClassic object is null

DamianoBarone, 2022-02-14:
**Mobile Device Environment**
Android tablet galaxy s7+

**Application Environment**
Provide information about your development environment:
- React Native version: [e.g. v5.0.1]
- RN Bluetooth Classic version: [latest, download today]
- Expo (latest)
- npm 8.5.0
- running on MacBook Pro 11.62

**Describe the bug**
I can't use any method of RNBluetoothClassic, it give me an error that the object is null
Error: [TypeError: null is not an object (evaluating 'this._nativeModule.isBluetoothEnabled')]
I have imported
 import RNBluetoothClassic from 'react-native-bluetooth-classic';
async function btList() {
  try {
    let devices = await RNBluetoothClassic.isBluetoothEnabled();
  } catch (err) {
    console.log('Error:', err);
  }
}
btList is called in the useEffect()  (I'm using function component)
I've installed the react-native-bluetooth-classic with npm and yarn, I have also tried the expo plugin. But same error.
In the main application.jave
    @Override
    protected List<ReactPackage> getPackages() {
      @SuppressWarnings("UnnecessaryLocalVariable")
      List<ReactPackage> packages = new PackageList(this).getPackages();
      // Packages that cannot be autolinked yet can be added manually here, for example:
      // packages.add(new MyReactNativePackage());
      packages.add(new MainReactPackage());
      packages.add(new ReactToolbarPackage());
      packages.add(new RNBluetoothClassicPackage());
      return packages;
    }
I added this, I tried different setting, but same  problem.

**To Reproduce**
Steps to reproduce the behaviour:
If I follow step by step your guide I'm ending in this problem

**Expected behaviour**
I should be able to call all the methods of RNBluetoothClassic
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
your job is incredible, I'm sad that I have to annoying you. 
I have also create this post 
https://stackoverflow.com/questions/71104150/null-is-not-an-object-evaluating-this-nativemodule-isbluetoothenabled-in-re

kenjdavidson, 2022-02-14:
You need to use an extra library to get this working with expo. I didn't write it but it was confirmed working by the author. Look in tee documentation for the expo section and maybe open an issue in his.

I don't use expo, and Bluetooth originally didn't work with expo, so I can't help much.

protected List getPackages() {
List packages = new PackageList(this).getPackages();

https://stackoverflow.com/questions/71104150/null-is-not-an-object-evaluating-this-nativemodule-isbluetoothenabled-in-re<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fstackoverflow.com%2Fquestions%2F71104150%2Fnull-is-not-an-object-evaluating-this-nativemodule-isbluetoothenabled-in-re&data=04%7C01%7C%7C5837666edb084c89ecd208d9ef4ec2e1%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637803943816930253%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=8aMBwOt0wnAuGixZqrncRFOE%2B2WGBohqYX0KED96LBE%3D&reserved=0>

DamianoBarone, 2022-02-14:
thank you very much for the quick answer.
Honestly I'm not dependent of EXPO, so I can choose anything else. Important is that bluetooth is working.
What do you use or recommend for you API?

kenjdavidson, 2022-02-14:
I just use the regular react native app. Expo wasn't good when I started the Bluetooth thing so I never bothered. It should work not with the provided library.

https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/124

DamianoBarone, 2022-02-14:
Ok thank you very much!!

closed 2022-02-14 by DamianoBarone