kenjdavidson/react-native-bluetooth-classic #169: Device freezes on to many messages

brunovollmer, 2022-02-15:
Hey,

before I start this I want to mention that I really like this library and that I'm not a professional JS/React developer so it might be that I'm just using some concepts wrong.

## System

**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: API Level 30 (Android 11)

**Application Environment**
Provide information about your development environment:
- React Native version: v0.65.1
- RN Bluetooth Classic version: 1.60.0-rc.19

## Problem

In our setup we have a python client communicating with our react native app via bluetooth. So far everything works fine. I can connect, send messages and reply. In some situations it can happen that the python client sends a lot more messages and this leads to the app losing all responsiveness (buttons don't work, etc.). 

My question would be is there a way to handle this inside react native as I guess the problem is somehow related to some event handlers or is the only solution to send less messages? As mentioned before I might miss some basic understanding of React here.

I've appended the code I think is relevant but if you need more information I'm happy to update the information.

Thanks in advance,

Bruno 

## Code

```JavaScript
const onDeviceDisconnect = async (
  device: any,
  readSubscription: any,
  dispatch: any,
  disconnectSubscription: any,
  resetNavigation: any
) => {
  readSubscription.remove();
  disconnectSubscription.remove();

  dispatch(getSelectedDevice(undefined));

  resetNavigation();

  console.log('disconnected device: ' + device.name);
};

export const connectingToDevice = async (
  device: any,
  onDataReceived: any,
  dispatch: any,
  resetNavigation: any
) => {
  let isConnected = await device.isConnected();

  if (!isConnected) {
    try {
      let connection = await device.connect({
        DELIMITER: '\n',
      });
      console.log('device connected successfully >>> ', connection);

      let readSubscription = device.onDataReceived((data: any) =>
        onDataReceived(data)
      );

      let disconnectSubscription = RNBluetoothClassic.onDeviceDisconnected(
        (device) =>
          onDeviceDisconnect(
            device,
            readSubscription,
            dispatch,
            disconnectSubscription,
            resetNavigation
          )
      );

      return connection;
    } catch (error) {
      console.log(error);
      return false;
    }
  } else {
    return isConnected;
  }
};
```

kenjdavidson, 2022-02-15:
I'm aware of this issue on ios, it's due to me not programming the stream interaction well. But on Android I've not seen this before. How many messages are you sending? What's the throughput.

The native code should be handling this OK, it's running a separate thread. But what's your listener doing? OnDataReceived? If that's doing too much work for each request you'll have issues.

Can you confirm whether it's react or Android that's having this issue? I don't have a device available to simulate this, none of my devices stream data.

If react has a constant stream, then you'll need to

- limit on python, although not the best option
- limit in js

For the latter there are many options, but it all depends on what data you have.

- are drops of data acceptable?
- do they need to be managed differently?
- etc

I can't answer this for you, they sre design issues.

If it's the native side, you may have to write a custom device connection native class that can handle the constant stream better.

kenjdavidson, 2022-02-15:
With regards to confirming `Android/Native` isn't the issue, you can try to connect without setting up the auto read.  This way you will have messages coming into the buffer on Android, but not hitting the UI Thread on React (js).  If this doesn't halt the app, then the Android side should be fine.

With regards to the limiting things on the React (js) side, you'll have to do some research on what's available to offload processing from the UI Thread on Javascript/React.   You need to think of getting getting bluetooth data much like getting HTTP data.  If you're running a reducer getting HTTP data then you're going to see the same issues.  There are a number of sites/libraries that document how to do Async reducers, this is probably what you're going to want to look into. 

If you're already doing this, then I'm not sure.  I'd have to spend way more time researching, time of which I don't have.

brunovollmer, 2022-02-17:
Hey @kenjdavidson,

thanks for the tips. I will spent some time in this direction and will let you know if it is successful or there are open questions.

kenjdavidson, 2022-02-17:
Sounds good. I'll leave the task open for a bit. Please update it if you get things resolved or figure something out.

closed 2023-03-25 by kenjdavidson