kenjdavidson/react-native-bluetooth-classic #170: Doubt about MFi

Luanfv, 2022-02-22:
Hello, I'm working on an application that communicates with a raspberry via bluetooth, this raspberry product is being developed by another company. That said, I'm having trouble making bluetooth communication through iOS, I saw several issues commenting on the MFi, but I would like to know if it's possible to at least do tests with MFi package tests (found on the internet) before asking the company sign up for apple, because today I'm not even able to list the paired devices.

Note: the app is already perfect on android

kenjdavidson, 2022-02-22:
If you can find an MFi device with free protocol strings, then yes you can test thing.  From what I’ve seen people experiement with, there are:

  *   Barcode scanners
  *   Printers
  *   Lego has a set that has them

You’ll have to search around though, I don’t know the specifics.

closed 2022-02-23 by kenjdavidson

Luanfv, 2022-02-23:
when I run the getBondedDevices function on iOS it returns an empty object, do you know what it can be?

Luanfv, 2022-02-23:
Correct me if I'm wrong, please, to perform this function I need the app and the device I want to connect to have the same package to work, right? If so, is there any other way to establish a connection without having to register this package on the device I want to connect to?

reference: https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/82#issuecomment-799614528

kenjdavidson, 2022-02-23:
If you're working with IOS the call to `getBondedDevices` will only return a list of paired devices for which you have the correct protocol Strings.   So, if you have a device paired it will show up in the IOS System list of devices, but if you do not have the correct protocol string, you will not see that in the `getBondedDevices` list.  This is working as defined by the `ExternalAccessory` documentation.

I don't know what you mean by **need the app and device to have the same package**?  You mean **protocol string**?

Luanfv, 2022-02-23:
I'm aware of very low iOS, so I apologize for the difficulty of expressing myself, but yes, I'm referring to ExternalAccessory, is there any way to establish a connection without them sharing the same?

kenjdavidson, 2022-02-23:
No, that's the whole purpose of the ExternalAccessory and MFi requirements.

Luanfv, 2022-02-23:
If i ask the people who developed the raspberry part to add the test mfi i found on the internet, would it work? thinking about tests, of course

kenjdavidson, 2022-02-23:
They can't just add a random protocol string.  They need to pay a bunch of money and sign a bunch of NDA's and go through a bunch of testing to get their device on the MFi program.  If they are willing to go through that process, and provide you with this string, then yes you will be able to communicate.

When working with the MFi program, for every version of the application you release, you need to reach out to that company and have them submit a new request to the MFi program stating that your new application version is still permitted to use those protocols.

The whole process is crazy annoying, and I would suggest you move to BLE if it's remotely possible.

kenjdavidson, 2022-02-23:
For testing, once they are on the MFi list, you can use their protocol strings without having your application validated by Apple.  But once you're ready to go live, you have to jump through all the hoops.

Luanfv, 2022-02-23:
what is the difference between bluetooth classic and ble?

kenjdavidson, 2022-02-23:
https://medium.com/@akash.kandhare/bluetooth-vs-bluetooth-low-energy-whats-the-difference-74687afcedb1#:~:text=Just%20like%20Bluetooth%2C%20BLE%20operates,Bluetooth%20which%20would%20take%20~100ms.

kenjdavidson, 2022-02-23:
With regards to IOS, BLE doesn't use the ExternalAccessory library, it uses CoreBLuetooth, which doesn't require any of the MFi nonesense.