kenjdavidson/react-native-bluetooth-classic #173: onDeviceConnected is not triggered?

ShaneZhengNZ, 2022-03-05:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Samsung Galaxy M22
- OS: Android 11

**Application Environment**
Provide information about your development environment:
- Expo: v44.0.0
- React Native version: 0.64.3
- RN Bluetooth Classic version: 1.60.0-rc.20

**Describe the bug**
It looks like the RNBluetoothClassic.onDeviceConnected event is not triggered when the Bluetooth device is connected.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an Expo RN project with Bluetooth feature enabled.
2. Add code to subscribe to RNBluetoothClassic.onDeviceConnected and RNBluetoothClassic.onDeviceDisconnected
3. Try to connect and disconnect a paired Bluetooth device. I am using SONY wireless headphones (WF-1000XM3).
4. When the headphone is connected, the onDeviceConnected callback is not called, however, when I switch the headphone off (device disconnected), the onDeviceDisconnected callback is called.
5. I also tried to connect and disconnect programmatically with .connect() and .disconnect() methods. Exactly the same issue, onDeviceConnected does not work, but onDeviceDisconnected works.

**Expected behavior**
RNBluetoothClassic.onDeviceConnected event should be triggered when paired device is connected to the phone.

kenjdavidson, 2022-03-05:
It looks like I removed it from the acl receiver on purpose. Can't remember why but I'm guessing that when I converted all the requests to promises it made no sense to lisen for a connection as the result of the promise should give you want you need, or reject if failed.

Disconnect is needed so you can handle cleanup appropriately.

Is there any reason why you'd prefer the event over the promise resolve?

kenjdavidson, 2022-03-05:
Can easily add it back into the AclReciever and Implement in the module. Feel free to give it a go and submit a pr if you confirm it working. 

But again using. Connect should return a promise with he device. Also headphones are usually not rfcomm, they usually need the a2dp profile (from other issues ppl have asked about).

ShaneZhengNZ, 2022-03-06:
I didn't try the promise, will try that first. It that works, then the onDeviceConnected event is not required. In which case, I totally agree that the js and documentation should be removed. Thank you for the prompt reply

kenjdavidson, 2022-03-06:
Keep me Posted bud.

ShaneZhengNZ, 2022-03-06:
I am sorry, in fact, I am using the promise.

Here is what I am doing - await the promise to resolve with an infinite loop to achieve auto connect
```javascript
let connectionEstablished = false;
while (!connectionEstablished) {
  try {
    connectionEstablished = await deviceToConnect.connect();
    if (connectionEstablished) {
      // Run my device connected function
      break;
    }
  } catch (error) {
    // Do nothing when the connect fails. Just back to the loop and connect again.
  }
}
```

I am not 100% happy about the infinite loop, but I don't think the onDeviceConnected event handle can solve this either. It is an ok solution for now (at least from my perspective), not sure whether you have any better way to achieve the similar auto connect feature.

If you agree, I can close this issue. Maybe open a PR to delete the onDeviceConnected event handle and documentation.

kenjdavidson, 2022-03-06:
You shouldn't need on the loop. I think you're running into an issue with a headset needing a2dp and not rfcomm. I don't think you can use this library for headsets. There are a few issue open and closed already about this.

ShaneZhengNZ, 2022-03-06:
I don't know too much details about bluetooth, that's why I use a headset as an example, obviously it is not appropriate.

I am actually using the library to connect to Bluetooth GPS device (Garmin GLO™ 2). Not sure whether the loop is appropriate to achieve auto connect? If not, what is your suggestion?

kenjdavidson, 2022-03-06:
I'd say no. Either the device connects or it doesn't. There is no point in a loop, as you have to handle the exception of fails.

ShaneZhengNZ, 2022-03-06:
True, however, that means I can't achieve auto connect. With the loop, I can have the loop running so that whenever the Bluetooth device is powered on, it connects.

Unless there are other ways to achieve that auto connect that I am not aware of.

kenjdavidson, 2022-03-06:
That's not what connected is anyhow. Connected is the rfcomm socket. There is no way to know when the device comes online besides try/catch on connect like you're doing in a loop.

But it should be try catch, not try false.

closed 2022-07-28 by kenjdavidson

RahulMvr2824, 2023-03-23:
Hi kenjdavidson,
I'm designing an App where device takes the sensor readings from hardware model and displays it on multiple screens which is the requirement for us. So in order to do that I had to perform two communication. Therefore, I need to check the Bluetooth connection status and pass the message from the mobile to desired hardware model. 
I was facing trouble in handling onDeviceConnected. Could please guide me how to resolve that.

kenjdavidson, 2023-03-23:
On device connected was deprecated. It was an old method that posed no real purpose after i updated the library. You get a device back from the connection request now that should be used. 

You'd have to provide explicit details to your use case, as there are other ways to handle it.

RahulMvr2824, 2023-03-24:
My use case is I'm building an app with multiple screens like home page, temperature measurement page etc., including Bluetooth connectivity page which means for connecting Bluetooth devices I'm using a separate page. So here I'm navigating from home page to Bluetooth page and connecting to the device required and coming back to home page where my two way Bluetooth communication starts. From home page itself I need to read data from the hardware module and write data to module. So here in need to check the device Bluetooth connectivity first.

kenjdavidson, 2023-03-24:
Ya. That's not what on connected device was originally for. So even if it 2asnt deprecated it wouldn't work how you're expecting.

You just want to connect to a device and store that connection at the app level. Just remember that lifecycle changes will kill the connection.

RahulMvr2824, 2023-03-25:
Here is my another query. In my application I need to send the integer value to the hardware module. But it's not accepting the number. Help me out to find a way to send integer values.
Thanks in advance.

RahulMvr2824, 2023-03-31:
Hi there,
I working on a react native project for IoT device where the sensor readings from the hardware needs to be displayed on Mobile App through Bluetooth. Here I'm very well established a connection to the device and also able to communicate with the device. 
But the problem is I cannot able to read the data at the same instant the hardware device throws it into the channel. Which means I'm missing out some data in between two read operations. Where as while read through the for loop as mentioned in your repository code it's getting lagged between the data which comes into channel and data which is fetched.(for better understanding, the instant I fetch one data value there are tons of data getting logged into buffer).

So kindly please help me with this kind of issue on smooth & efficient data communication.
Thanks in advance

kenjdavidson, 2023-03-31:
`But the problem is I cannot able to read the data at the same instant the hardware device throws it into the channel`

Are you not using the `onDeviceRead` handler, which should be passing messages to your React side code as soon as a delimited message is found?  If this is a separate issue, you should:

- open a new issue
- provide actual code samples
- give as much information as possible

from what you're saying I can only imply that you are only doing `timeout`/`polling` type `device.read()` and you are missing messages?  Which still shouldn't be the case, since `device.read()` should eventually give you all the messages.

kenjdavidson, 2023-03-31:
You're going to have to deal with getting data on your own, you're two options are:

1. `onDeviceRead()` which is event based
2. `read()` which is polling based

both have their pros and cons.

You can even go so far as writing your own `BluetoothConnection` implementation and handle direct data that way.  The library should be open enough for any number of options.

RahulMvr2824, 2023-04-01:
> You're going to have to deal with getting data on your own, you're two options are:
>
> 1. `onDeviceRead()` which is event based
> 2. `read()` which is polling based
>
> both have their pros and cons.
>
> You [...]

Thank you for your help. Will try the same and post the outcome.