kenjdavidson/react-native-bluetooth-classic #175: Option to access device type

brunovollmer, 2022-03-14:
**Mobile Device Environment**
This feature would be needed on IOS as well as Android.

**Is your feature request related to a problem? Please describe.**
I would like to know what type of paired or unpaired Bluetooth device is. More specifically I would like to know if I can play audio over this device. This should in theory be possible as Ubuntu offers that functionality in their Bluetooth menu.

**Describe the solution you'd like**
I don't know the internals of this library too well but I guess the best place to store such information would be inside the Bluetooth device object.

kenjdavidson, 2022-03-14:
> I just checked the code. Is deviceClass what I'm looking for? And if yes, what could be the reason that is is always undefined?

I would assume yes. If device class is null then it wasn't provided by the underlying Bluetooth adapter. Both android and ios attempt to set this if provided.

brunovollmer, 2022-03-14:
So I've checked. Ubuntu does detect that my headphones are headphones. Any idea where the problem could be and where I could have a look?

kenjdavidson, 2022-03-14:
You've got two choices.

Discover devices on discover where the Bluetooth devi e is mapped for react.

Paired devices where they are also mapped.

In both cases it should be attempting to set device Class.

I tried to map all the available Android Bluetooth device fields, then match those on ios. I can't imagine they've added more since Bluetooth isn't new.

I'd start in that native code and see what the values are on Android.

brunovollmer, 2022-03-14:
Ok thanks for the detailed answer. One last question. How would you recommend to debug a problem like this? Where can I find the Android code that is installed? Inside the npm package?