kenjdavidson/react-native-bluetooth-classic #180: unpairDevice function don't works

jojemapa, 2022-04-16:
**Mobile Device Environment**
- Device: [Android HUAWEI Y7 2019]
- OS: [API27, Oreo]

**Application Environment**
- React Native version: [v0.67.2]
- RN Bluetooth Classic version: [v1.60.0-rc.20]

**Describe the bug**
I call the 'unpairDevice' function and it returns 'true' but it doesn't do anything, I even checked the bluetooth settings and the device is still paired.

**To Reproduce**
Make a simple app, install and configure the library, and call the function unpairDevice.

`const gotUnpaired = await RNBluetoothClassic.unpairDevice(address);`

**Expected behavior**
This should unpair the device, so that calling the 'RNBluetoothClassic.getBondedDevices()' function no longer returns it, and even checking the bluetooth settings doesn't show it.

I'm sure the problem is the unpairDevice function, since I manually unpaired it from the phone's bluetooth settings screen and the 'RNBluetoothClassic.getBondedDevices()' function no longer returned it.

**Screenshots**
I'm testing React Native and i am making a bluetooth app for funny.
[image]
_I don't know why I cover this device with red, but I read a book that even by the sound of the keys you press on the keyboard they can tell what you write 😲, and well._

**Additional context**
Although this functionality is not required in my app, and as the documentation says 

> Note I haven't actually used or spent any time testing this. It was a holdover from the original library. Feel free to update me with information on whether this is working.

I think it could be cool to have this like this to handle everything well, although my goal is to see if I can send a command to a printer here, to see what comes out 😊

kenjdavidson, 2022-04-16:
I completely agree. I'll see if I can take a look at it at some point. But if anyone happens to submit a pr, I won't be disappointed.

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F22124150%2F163656654-611a6aba-00c0-4f72-8e61-0b79a933ba1d.png&data=04%7C01%7C%7Cffb072b7470744fb7f3c08da1f492ce6%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637856696390622724%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=%2BLB3xO8%2Bl4mpYOUuDnUzvix%2F9T5v7tkhTKil97B9GrM%3D&reserved=0>

jojemapa, 2022-04-17:
The fix is easy, i think. error of copy paste. 

Line 593, File RNBluetoothClassicModule

Change this 
Method m = device.getClass().getMethod("createBond", (Class[]) null);

To
Method m = device.getClass().getMethod("removeBond", (Class[]) null);

kenjdavidson, 2022-04-17:
That'll happen. Thanks for looking at it.

kenjdavidson, 2022-04-17:
Released.

closed 2022-04-17 by kenjdavidson