kenjdavidson/react-native-bluetooth-classic #184: How to read a hex sent from a device

abdulrafeh857, 2022-05-13:
I am sending a hex value to a device and expected to receive hex value in response `08 00 00 05 01 00 00 01 03 00 0E`. But i am unable to do so. The object that comes in onDataReceived has the key data with empty value. Please guide how to read hex data from a device

abdulrafeh857, 2022-05-13:
Write Function
  let writeData = await RNBluetoothClassic.writeToDevice(
        props.device.address,
        message,
        "hex"
      );

kenjdavidson, 2022-05-13:
When you send the hex value to the device, are you getting the hex data on the device? 
It's just when you're sending hex data back from the device that you're getting nothing?
Are you sending a delimiter with the data from the device? Without a delimiter (which isn't isn't usually the standard with byte data, it's usually more exact byte sizes) the messages will never be split and returned. 
What happens when you call the manual `read` method after sending data from the device?

abdulrafeh857, 2022-05-13:
First of all, thank you so much for this awesome library. 

When you send the hex value to the device, are you getting the hex data on the device?
Yes

It's just when you're sending hex data back from the device that you're getting nothing?
Yes i have an issue with read data.

Are you sending a delimiter with the data from the device?
That device is actually a meter, so i don't think it sends any delimiter.

What happens when you call the manual read method after sending data from the device?
i also tried the manual read method, with a setInterval. The .available method shows 11 available, but consoles nothing.

kenjdavidson, 2022-05-13:
So ya, if you're not sending a delimiter `\n` or whatever, then the messages will never be broken up, and therefore never be found.  What it does is:

- Data comes in
- Do we have any delimiters?
- For each delimiter send that data (if there is a listener)

for manual reads
- Do we have a delimter?
- Each read will get the next data for the delimiter

If you're getting 11 available, that implies that there should be 11 separate messages, I'm pretty sure that's what it means anyhow.

You can attempt to do

```
connectToDevice.(deviceId, { delimiter: '' });
```

Using an empty/null delimiter, MAY (I think there was a request for this) cause it to skip all the delimiter logic, and just send the stream of data through.

abdulrafeh857, 2022-05-16:
Thank you @kenjdavidson . I solved my problem from this 
https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/50#issuecomment-702030437

closed 2022-05-16 by abdulrafeh857