kenjdavidson/react-native-bluetooth-classic #190: Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1

yanwzhang, 2022-06-27:
=
- Device: Android 11
- React Native version: 0.64.1
- RN Bluetooth Classic version:1.60.0-rc.21

**Describe the bug**
when i try to connect device, it throw error:  java.io.IOException: read failed, socket might closed or timeout, read ret: -1

the follow is my code: 

 const [devices, setDevices] = useState([]);

  const getBondedDevices = async unloading => {
    try {
      let bonded = await RNBluetoothClassic.getBondedDevices();
      console.log('DeviceListScreen::getBondedDevices found', bonded);
      setDevices(bonded);
    } catch (error) {
      console.log('get devices error:', error);
    }
  };

const connect = async device => {
    try {
      let connection = await device.isConnected();
      console.log('是否连接：', connection);
      // 未连接
      if (!connection) {
        await device.connect({delimiter: '\r'});
        console.log('连接结果：', connection);
      } else {
        device.disconnect();
      }
    } catch (error) {
      console.log('连接过程错误：', error);
    }
  };

const renderItem = device => {
    return (
      <TouchableOpacity onPress={() => connect(device)}>
        <View style={[styles.item]}>
          <View>
            <Text style={styles.name}>{device.name}</Text>
            <Text style={styles.itemId}>{device.address}</Text>
          </View>
        </View>
      </TouchableOpacity>
    );
  };

<FlatList
        data={devices}
        renderItem={({item}) => renderItem(item)}
        keyExtractor={item => item.address}
      />

kenjdavidson, 2022-06-27:
Hey, need a little more details on what you're attemtping to do:

1. What type of device are you attempting to connect with?
2. Can you confirm that the other device is actively waiting for a connection? If you're doing this manually you need to ensure a BluetoothServerSocket has been started.

Othwerise you may need to perform some debugging yourself and supply the results here.  It's very difficult to assist with these errors are they are generally very device/custom code specific.

yanwzhang, 2022-06-28:
Thank you for your reply.
1. I am attempting to connnect a bluetooth speaker which can dicovery with the function startDiscovery.
2. I confirm that my bluetooth speaker is waitting for a connection.
3. I also try to copy the react-native-bluetooth-classic-app demo, it run with the same error.

yanwzhang, 2022-06-28:
can you give me the right step process for a connection between phone and a bluetooth device, thank you very much.

kenjdavidson, 2022-06-28:
Sure, for the most part (at a high level) you need to:

1. Scan for the device
2. Pair the device 
3. Turn the device on 
4. Turn the device into paring mode (some are automatic, some need to be done manually).  Since I have no idea what device you're using I can't give more details.
5. `device.connect()`

This should get you connected.  If you're getting the connection error you are, I would expect that:

1. The device isn't on (hopefully unlikeliy)
2. The device isn't in pairing mode 
3. The device has another device connected?  Some devices will automatically drop when a new request comes in, some will not.

Bluetooth speakers don't generally allow for RFCOMM communication, they either use:
- L2CAP (which this library doesn't support)
- The Bluetooth Audio Profile (which this library doesn't support)

So you need to make absolutely sure that your speaker supports the RFCOMM connection method.  Which is unlikely.  You'll find some other attempts to connect to things like:
- speakers
- headsets
- etc

which have not been possible.

closed 2022-06-28 by kenjdavidson