kenjdavidson/react-native-bluetooth-classic #196: Issue reading from OBD II device

amireds, 2022-08-04:
- Device: Android

**Application Environment**
- React Native version: 0.68.2
- RN Bluetooth Classic version: ^1.60.0-rc.23

**Bug Description**
Hi! Awesome work done on this package. My goal for using this package is to actually allow my app talk to an OBD II device that uses classic Bluetooth.
I was able to connect to the device seamlessly - your tutorials were really helpful.

However, in trying to read data from the device, I keep getting null, I've tried several directions, both reading using `onDataReceived`  and using `.read()`
None seems to be working.

When I tried writing to it, I get a true response. but read keeps giving me `null`.
Please help.

Below is my function snippet

**To Reproduce**
```
async function readConnected() {
    setReading(true);
    const connected = await device.isConnected();

    if (connected) {
      try {
        let writeMessage = await device.write("AT SP 0");
        console.log("Write message: ", writeMessage);
        if (writeMessage) {
          //let message = await device.read();
          //console.log("Read message: ", message);
          let subscription = await device.onDataReceived((data) =>
            onReceivedData(data)
          );
          //console.log("Received data: ", message);
          //console.log("Subscription: ", await subscription.remove());
        }
      } catch (error) {
        // Handle error accordingly
        console.log("Error reading message: ", error);
      }
    }
    setReading(false);
  }
```

**Screenshots**
I added a screenshot just to see what has been done
[image]

amireds, 2022-08-04:
Hi, @kenjdavidson  Thanks for the response. Below is the code for the connection 

```
async function connectToDevice() {
    setIsConnected(false);
    setConnecting(true);
    try {
      if (device._nativeDevice.bonded === false) {
        console.log("Bonding to device");
        const paired = await RNBluetoothClassic.pairDevice(device.address);
        setDevice(paired);
      }
      console.log("Connecting to device");
      const checkConnected = await device.isConnected();
      if (!checkConnected) {
        const connected = await device.connect({
          CONNECTOR_TYPE: "rfcomm",
          DELIMITER: "\n",
          //DEVICE_CHARSET: "ascii",
          DEVICE_TIMEOUT: 10000,
          SECURE_SOCKET: false,
        });
        console.log("Connected to device", connected);
      }
      setIsConnected(true);
    } catch (e) {
      console.log("Error in connection: ", e);
    }
    setConnecting(false);
  }
```

Also for the documentation... Would these suffice?

https://www.scantool.net/scantool/downloads/234/stn1100-frpm-preliminary.pdf
http://www.obdtester.com/elm-usb-commands

amireds, 2022-08-04:
Is there a possibility of a possible virtual meet, If you don't mind, we could use google meet?

kenjdavidson, 2022-08-04:
Sure, anything is possible!!

But this week is out, so maybe one evening (EST) next week I could have some time.  I just got back from Vacation and I'm going away again this weekend.

amireds, 2022-08-04:
Oh Great! Thanks a lot.
Pending that time, I'll appreciate a guide through the debugging.  What do you need me to do, let me try it here.

kenjdavidson, 2022-08-04:
React Native will pull all the released source code into the Libraries folder of Android Studio (or the Pods folder of XCode).  So there isn't much needed, you just need to open your application in Android Studio (you do this by selecting the `/android` folder of the project, then open the dependency file within that same session and set a breakpoint.

So I guess the only thing is Android Studio that you'll need.

kenjdavidson, 2022-08-17:
Hey sorry, work and fun have been busy this last little while.  Have you made any headway with this?  I might be able to spend some time in the coming week.   I'd prefer if you've been able to get it debuggin in android studio before we worry about a screen share.

Let me know.

kenjdavidson, 2022-09-08:
Is this still an issue?

closed 2022-09-26 by kenjdavidson