kenjdavidson/react-native-bluetooth-classic #200: Why I cant fix a timeout to startDiscovery like in Bluetooth Serial ?

geralda05, 2022-09-09:
**Mobile Device Environment**
Android

**Is your feature request related to a problem? Please describe.**
Yes, I'm trying to scan available devices but it's very slow. It takes 2 minutes approximately.

**Describe the solution you'd like**
Set a timeout option for startDiscovery (in bluetooth serial already exists) or a onDeviceDiscovered function

kenjdavidson, 2022-09-09:
All this does is pass through to the android system. You can easily add your own time out to stop scanning based on your own rules to cancel discovery.

Not sure whether it's worth adding into the library at this point in time.

geralda05, 2022-09-09:
Thanks. Is there a function like "onDeviceDiscovered" for show available devices while the scan is running ?

geralda05, 2022-09-09:
I closed it by mistake. Sorry. Is there a function like "onDeviceDiscovered" for show available devices while the scan is running ?

kenjdavidson, 2022-09-09:
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/5a9b0c2249da025bec431e7f4f47bcdf8c049997/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java#L452

Does that do it?

kenjdavidson, 2022-10-11:
Any update on this, did you get the `DEVICE_FOUND` event working?

closed 2022-11-29 by kenjdavidson

seba9999, 2024-02-28:
Hello, for the record I've tried this event and it doesn't trigger ... Even though my promise from startScan return an element ... :/ 

Has this event been deprecated ?

kenjdavidson, 2024-02-28:
Not on purpose.  If it's not working then feel free to PR a fix for it.  It should be there since it's in the Native code, unless something on the JS side is causing it not to get passed through.