kenjdavidson/react-native-bluetooth-classic #202: Error building APK

geralda05, 2022-09-26:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android - compileSdkVersion 31

**Application Environment**
Provide information about your development environment:
- React Native 0.7 with RN Bluetooth Classic v1.60.0

**Describe the bug**
I'm trying to create an APK of my react-native app running assembleRelease and i'm getting this error at the end of the command:
**"> Task :react-native-bluetooth-classic:verifyReleaseResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-bluetooth-classic:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.tasks.VerifyLibraryResourcesTask$Action
   > Android resource linking failed
     ERROR:C:\Users\Tecnoandina\.gradle\caches\transforms-3\efc5fbdc5cddabff1e6dbe6ad771c56d\transformed\core-1.7.0\res\values\values.xml:105:5-114:25: AAPT: error: resource android:attr/lStar not found."**

Any suggestion? What must I do? :(

kenjdavidson, 2022-09-26:
I'm assuming that the library isn't compatible with:
- React Native 0.7
- And because of that Android 31

May need to bump up to `react-native-bluetooth-classic v0.70.0` to take advantage of React Native 0.7 and update the Android min version to match.

If you can confirm this is required, I will:
- Create the `releases/0.60` branch
- Bump `main` to `0.70` and accept a PR

geralda05, 2022-09-26:
> I'm assuming that the library isn't compatible with:
>
> * React Native 0.7
> * And because of that Android 31
>
> May need to bump up to `react-native-bluetooth-classic v0.70.0` to take advantage of React [...]

Yes, I'm using react native 0.7 and android 31 sdk. Thanks for help me !

fdrault, 2022-09-27:
Not sure why, but I have met this exact same error with others packages.
The problem was some packages building with different `buildToolsVersion` and `compileSdkVersion` from my app's.

I created a PR (#203). This way Gradle grabs versions defined in your project version and use it, but will fallback on default package version if not present, exactly the same way that `react-native-reanimated` does
https://github.com/software-mansion/react-native-reanimated/blob/main/android/build.gradle#L241

It should fix the problem

geralda05, 2022-09-27:
Thanks for the PR!! It works now <3

PR #203 (2022-09-27): fix(build): Use project global buildToolsVersion and compileSdkVersion
Should fix #202 

Make gradle use versions defined by your project 
* `compileSdkVersion` 
* `buildToolsVersion`
* `minSdkVersion`
* `targetSdkVersion`

They are grabbed from your global `android/build.gradle` and should automatically evolve with react-native updates