kenjdavidson/react-native-bluetooth-classic #204: Problem with reading data from bluetooth classic device(SPP)

nebojsa-salai, 2022-10-07:
**Mobile Device Environment**
- Device: Android
- OS: v12

**Application Environment**
- React Native version: 0.69.3
- RN Bluetooth Classic version: latest

**Describe the bug**
We have a problem with reading data from bluetooth classic device. For establishing a connection to the device we use method **connectToDevice**, passing two parameters, MAC address of device and json object which represents specification for serial communication( **{ connectorType: 'rfcomm', secureSocket: true, readTimeout: 3 * 1000, delimiter: '\n' }** ). Writing commands to the device was successfully completed but we did not receive any data from the device. We tried reading the data in three different ways:
    1. Listen received data through listener of device( **device.onDataReceived** )
    2. Listen received data through listener of RNBluetoothClassic( **RNBluetoothClassic.onDeviceRead** )
    3. Read data after command is sent( **device.read** )
    
We know that the Bluetooth SPP device works fine because we track logs on the device after the command has been sent to device.

kenjdavidson, 2022-10-07:
If you're confident that the device is sending data, then the next steps are:

1. Debug the `DelimitedStringCommDeviceConnectionImpl#onReceive` (I think) method to ensure that the data is in fact being received.
2. Confirm that the delimiter is correct, this is most likely the issue (in 90% of cases)

You're using `\n` which is the default, but are you sure it's not `\r\n`?  
Which device are you using? In house? Do you have documentation on the API?

What happens when the delimiter is not correct:
- data is received
- data is NOT parsed into delimited bundles
- whole content is added to the buffer

When this occurs:
- `onDataReceive` will never be called, since the message handler doesn't think there are any delimited messages
- `available` will return `0` since there are no delimiters found, and therefore `read` will return no data

nebojsa-salai, 2022-10-10:
I set appropriate delimiter and now I can see data from device. Now I have two problems:
   1. device.onDataReceive doesn't  always show data(on device logs data is shown after every command)
   2. Data must be hex but if I try to set hex encoding on connect method, connection to device is not possible

kenjdavidson, 2022-10-10:
There are a number of documents/questions here regarding hex data. It's rather annoying but doable. This is a react native limitation.

With regards to your first comments, I'd need to see more details. If the log shows the message, but Javascript isn't getting it, then I'm assuming something is missing in your code. I'd need to see examples.