kenjdavidson/react-native-bluetooth-classic #205: how to know bluetooth device connected to app

taehyun-kim-tenuto, 2022-10-11:
**Mobile Device Environment**
On which device(s) would you like this feature to be implemented.   If Android only, the IOS implementations must be created as well - keep not of this for your pull requests.
- Device: [e.g. Android, iPhone6]
- OS: [e.g. API19, Nouget, iOS8.1]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.  If you've researched the available libraries, feel free to add notes or implementation details as you see fit.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

i just read onDeviceConnected removed
that was exactly what i want
what should i do??

kenjdavidson, 2022-10-11:
Could you provide more information? I don't understand what you're asking.

taehyun-kim-tenuto, 2022-10-11:
I just want to know when the Bluetooth device is connected in the app
just exactly opposite side of onDeviceDisconnected

kenjdavidson, 2022-10-11:
First you need to understand the definitions of the functions and how bluetooth works.  With regards to this application:
- Paired = the bluetooth device becomes paired with the phone
- Connected = the bluetooth device and device successfully create an RFCOMM socket

The original `onDeviceConnected` referred to the RFCOMM socket being opened (this is why there is a PR to remove it).  It makes no sense because `await connectToDevice()` returns back when it's successful, or throws when it is not successful.

`onDeviceDisconnected` is required as the RFCOMM socket could be closed for any number of reasons.

taehyun-kim-tenuto, 2022-10-11:
thanks for your reply
I want an event listener or custom hook state value provided when the Bluetooth headset is connected or paired
i just want this If it's paired or connected, you can see the Bluetooth icon and If the pairing is broken, I want to show an icon that can't be bluetooth
What should I do to do that?

kenjdavidson, 2022-10-11:
Well:

1) There is no `BluetoothAdapter` functionality for listening for a paired/bonded device.  Therefore there is no way to provide this functionality to React Native.  The only way to do this would be to set a timeout that continually polls for bonded/paired devices and does something when it finds the device you're looking for.

2) Bluetooth headsets use the A2DP protocol, which this library is not designed for.  So there is no way within this library to wrap the functionality for an A2DP connection to be monitored.  The `BluetoothHeadset` is deep in Android and not something that this library will be responsible for.  If you wish to add your own custom `DeviceConnector` implementation then you can do so, but I don't see it as being part of this specific library.

3) `onDeviceDisconnected` is only fired when a device is disconnected, ie. the RFCOMM socket is broken by anything that didn't call `device.disconnect()` 

There have been a number of discussions about Headsets, and I'll be honest that's not what I wrote this library for, nor plan on really implementing directly.

kenjdavidson, 2022-10-11:
I would suggest implementing a library that takes advantage of A2DP and the `BluetoothHeadset` protocol.

closed 2022-10-11 by kenjdavidson