kenjdavidson/react-native-bluetooth-classic #210: Always Get "need android.permission.BLUETOOTH_CONNECT" when using getBondedDevices

WongTampan123, 2022-11-25:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android12

**Application Environment**
Provide information about your development environment:
- React Native version: v0.70.6
- RN Bluetooth Classic version: v1.60.0-rc.24

**Describe the bug**

I Always got this error when trying to use getBondedDevices. I've edit AndroidManifest.xml manually but the error presist.

Is there something i need to configure?

This is the error:
[image]

mojjominion, 2022-11-25:
reference: #179 

I faced this issue a couple of days back.
I was able to get it working, by using  `AndroidPermission` to ask for required permission at runtime. 

```
import { Permission, PermissionsAndroid } from "react-native";

const permissions: Permission[] = [
  "android.permission.BLUETOOTH_SCAN",
  "android.permission.BLUETOOTH_CONNECT",
];

... 
const granted = await PermissionsAndroid.requestMultiple(permissions);
allGranted = permissions.every((p) => granted[p] === "granted");

```

kenjdavidson, 2022-11-25:
I believe this is a documented issue in Android versions. That the application must request access to these permissions for use. I know this is true for camera an dither things.

I'll update the docs. But I believe this is working as designed with Android.

WongTampan123, 2022-11-25:
Thank you for the replies.
I know that I have to request access to these permission, but I don't know how to do it... All the sources I read they just told me that I have to include the permission in AndroidManifest.xml, but not in the app

Once more, thank you now it works

closed 2022-11-25 by WongTampan123

kenjdavidson, 2022-11-25:
Added to `Getting Started with Android` section of the documentation site.