kenjdavidson/react-native-bluetooth-classic #214: function connectToDevice()

leandro-wrocha, 2022-11-26:
**Mobile Device Environment**
Android 9.0 - SAMSUNG J4  PLUS

**Application Environment**
- React Native version:  0.70.6
- RN Bluetooth Classic version: 1.60.0-rc.24

**Describe the bug**
Error when trying to make a bluetooth connection with another device, the error that occurs during the connection response is this:
[Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1]

It doesn't return anything, it just falls into try,catch and returns this error

**To Reproduce**

This is the project, just download and install dependencies and run project

https://github.com/leandro-wrocha/bluetooth-app

[image]
[image]

kenjdavidson, 2022-11-26:
A couple of things?

What device are you trying to connect to?
Why are you using both accept and connect? Putting this device into accept mode based on permission, are you using this on two phones?

Without knowing what you're trying to do, I can't give move direction. But I suggest reading the documentation and the sample app.

kenjdavidson, 2022-11-26:
There is a sample app that provides this functionality. You can look at that.the steps are. 

Install app on both phones 
Enable Bluetooth on both. 
Pair both 
Put one app in accept mode
Connect to device on the other. 

This works in the sample app. You can fork/Clone it and try it out.

kenjdavidson, 2022-11-29:
Confirmed up and running via email.

closed 2022-11-29 by kenjdavidson