kenjdavidson/react-native-bluetooth-classic #224: [Question] is there any additional settings for the Eyoyo device?

rzfzr, 2022-12-16:
I just bought the eyoyo bar reader device mentioned in the readme by @kenjdavidson 

Connected it via SPP mode to my android device, [this app](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en&gl=US) reads the information from it fine.

However my app with this module, connects fine, but the onDataReceived listener is never triggered, any pointers?

kenjdavidson, 2022-12-16:
The default delimiter needs to be changed. Either on the app or the scanner. By default it's \n and the other is \n\r.

rzfzr, 2022-12-16:
Perfect, I tried switched around reading the code at the page 19 of the manual, I think it needed to reconnect.
Thanks

closed 2022-12-16 by rzfzr