kenjdavidson/react-native-bluetooth-classic #225: Android 12 Crashing when using RNBluetoothClassic.startDiscovery()

geraldcells18, 2022-12-17:
Thank you for developing this library, if I can personally thank you I will do it but I'm living in the Philippines. 😅

**Mobile Device Environment**
- Device: Android
- OS: Android 12

When using the await RNBluetoothClassic.startDiscovery(); in a device  with Android 12 OS the app is crashing.

**Describe the solution you'd like**
This is what I need to do to avoid crashing. 
We need to update the library permission and add the following:
- android.permission.BLUETOOTH_SCAN
- android.permission.BLUETOOTH_CONNECT
- android.permission.android.permission.ACCESS_COARSE_LOCATION (For Android 10 below)

**Additional context**
I based my fixes here on the android documentation for Android 12
- https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

kenjdavidson, 2022-12-17:
You're saying that you'd like these added to the libraries manifest file?  I'd prefer to keep these out of the library as I don't want to specifically tie the library to permissions.  It should be the application developer that chooses which permissions are required.

geraldcells18, 2022-12-18:
@kenjdavidson,

I will note this,
Thanks for your response!

closed 2022-12-18 by kenjdavidson