kenjdavidson/react-native-bluetooth-classic #239: react-native-bluetooth-classic this library is not scan device properly its scan two or three time then scan device

ayanWappnet, 2023-03-03:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: [e.g. Android 12,10,8]
- 
**Application Environment**
Provide information about your development environment:
- React Native version: [e.g. v0.69.1]
- RN Bluetooth Classic version: [e.g. v1.60.0-rc.23]

**Describe the bug**
issue is two or more time scan device 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'  open app
2. Click on '....'  scan device button
3. no any device found

**Expected behavior**
A clear and concise description of what you expected to happen.
expeted only one time bluetooth device scan
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
here is problem:
import RNBluetoothClassic from 'react-native-bluetooth-classic';
try {
    RNBluetoothClassic.startDiscovery().then((unpaired) => {
        this.setState({ devices: unpaired });
        let index = devices.findIndex((d) => !d.bonded);
        if (index >= 0) {
            devices.splice(index, devices.length - index, ...unpaired);
        } else {
            devices.push(...unpaired);
        }
    });
} catch (error) {
    concole.log('error', error);
}

kenjdavidson, 2023-03-03:
So..

1. What app are you referring too?  Your own custom app that has a `...` button? You'll need to provided that code if you want me to look/debug it.
2. I don't see anything wrong with your code, what is actually happening? Are you getting an error? Or are you actually getting NULL back as `unpaired`?

All this does is forward the request through to the `BluetoothAdapter`, you'll need to debug the Android/native code to see what's happening in the listener.

From what you've provided, I can't really help more than this.

ayanWappnet, 2023-03-03:
@kenjdavidson its bluetooth app issue is not scanning bluetooth device when (device is android v10 its scan 12 version but 12 version not scan 10 version) and i scan 3 4 times then scan properly

kenjdavidson, 2023-03-03:
So you have:
- 1 phone that is android 10
- 1 phone that is android 12

and you're scanning on your 12 phone and it's not seeing the android 10 phone?  Do you find other devices that are available when they are turned on?

ayanWappnet, 2023-03-06:
yes other device also scan

kenjdavidson, 2023-03-06:
If you're scanning on both devices this won't work, one device needs to scan and one device needs to broadcast.  If you can debug the issue and it turns out to be a change to the `BluetoothAdapter` please let me know (will also accept a PR), but without more specific details it could be anything.

ayanWappnet, 2023-03-10:
@kenjdavidson its working fine when  i pairing bluetooth device and open a bluetooth window page and then device detect in other device.This flow i am follow right now thats working fine but any way for bluetooth paired window can i show from my app?

kenjdavidson, 2023-03-10:
The library should be able to pair with a basic device within the app.  The issue is the Android to Android pairing obviously requires some extra steps that aren't available in the library.  At this point there are a couple options:

1. do it in the settings
2. open a PR with the changes required to do it in the app
3. wait for me to look into it (which sadly won't be any time soon based on my time and limited devices for testing)

ayanWappnet, 2023-03-21:
i am testing on onepluseNord ce 2 lite(android 12) and other side samsung j8(android 10)

closed 2023-03-25 by kenjdavidson