kenjdavidson/react-native-bluetooth-classic #247: why the func-isBluetoothEnabled is always false, even thought the bluetooth is open and app has the permission of bluetooth

raintoway, 2023-03-27:
**Mobile Device Environment**
iphone12: actual device, ios 16
react-native: 0.68.0
react-native-bluetooth-classic: lastest 

and i can't detect the state'change of bluetooth with `onBluetoothEnabled`

kenjdavidson, 2023-03-27:
Can you provide code?
Did you enable permissions?
Did you provide the mfi protocols for your device? Ios requires protocols even if you're using just the status events.
Have you attempted debugging?

raintoway, 2023-03-30:
> Can you provide code? Did you enable permissions? Did you provide the mfi protocols for your device? Ios requires protocols even if you're using just the status events. Have you attempted debugging?

sorry, my fault