kenjdavidson/react-native-bluetooth-classic #250: onDevice Read() delay problem

RahulMvr2824, 2023-04-10:
You're going to have to deal with getting data on your own, you're two options are:

1. `onDeviceRead()` which is event based
2. `read()` which is polling based

both have their pros and cons.

You can even go so far as writing your own `BluetoothConnection` implementation and handle direct data that way.  The library should be open enough for any number of options.

_Originally posted by @kenjdavidson in https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/173#issuecomment-1492252792_
          
Hi there,
I have tried implementing the onDevice Read() calling for data fetching on bluetooth channel. It's working fine but the event is triggering with delay. Means the hardware is sending the data with 4milliseconds delay but the onDevcieRead() is triggering with 1second because of this i'm having lag in my next operations and also screen response is delayed.

RahulMvr2824, 2023-04-10:
> ```
>           You're going to have to deal with getting data on your own, you're two options are:
> ```
> 
> 1. `onDeviceRead()` which is event based
> 2. `read()` which is polling based
> 
> both have their pros and cons.
> 
> You can even go so far as writing your own `BluetoothConnection` implementation and handle direct data that way. The library should be open enough for any number of options.
> 
> _Originally posted by @kenjdavidson in [#173 (comment)](https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/173#issuecomment-1492252792)_
> 
> Hi there, I have tried implementing the onDevice Read() calling for data fetching on bluetooth channel. It's working fine but the event is triggering with delay. Means the hardware is sending the data with 4milliseconds delay but the onDevcieRead() is triggering with 1second because of this i'm having lag in my next operations and also screen response is delayed.

I have observed that the above problem persists when i use onRecievedData function calling in perform read function. But i'm not having any delay when it is commented but the graph is not updating in UI.
What does this onRecievedData is used for. What happens when we do not use that?

kenjdavidson, 2023-04-10:
`I have observed that the above problem persists when i use onRecievedData function calling in perform read function.`

Don't do this, the data event and manual read are mutually exclusive.  You want to either:
- listen for data messages
- poll for data messages

but you do not want to call `read` from the data event handler, this is bad as the data is gone.

Without seeing your code, I can't really tell what you're actually doing, nor why something would be going wrong.  But the fact that you're using both the even and the read lead me to believe you're not following the example code.

kenjdavidson, 2023-04-10:
`RNBluetoothClassic#onDeviceRead` sets up the event listener for a particular device connection so that any delimited messages found are sent to the React Native side in the handler.

`BluetoothDevice#onDataReceived` is just a pass through to the `RNBluetoothClassic#onDeviceRead` for that particular bluetooth device.

> So literally these do the same thing, it just depends on how you're managing the logic, either directly with `RNBluetoothClassic` or on the `BluetoothDevice`.  If you feel like the documentation doesn't explain this well enough, feel free to open a task and I will attempt to update it.  

`BluetoothDevice#read` manually reads data from the buffer (if any exist) and should be coupled with `BluetoothDevice#isAvailable`.

> This method should be used with a `setInterval` or some other polling method.

kenjdavidson, 2023-04-11:
Well you can't use direct setting of state, you have to use set state. You'll have to look into async state management. 

Honestly, you should be using native Android to do this if performance is important. React and Javascript aren't great for super performant apps. You'd get better performance using native, and having threading available.

closed 2023-04-17 by kenjdavidson