kenjdavidson/react-native-bluetooth-classic #256: If you disconnect and connect the device, have a two events with onDataReceived

ValentinOsvaldo, 2023-05-23:
I used the library for some months and I have this bug

When the user disconnect and connect the device, before that the ```onDeviceDisconnected``` listener detect the device and show the toast, we have a problem, the listener ```onDeviceRead``` (or ```onDataReceived```) will run twice.

This is my code to make the connection:

```ts
const connectDevice = (device: BluetoothDevice): Promise<void> => {
    return new Promise(async (resolve, reject) => {
      try {
        await device.connect({
          connectionType: 'binary',
          delimiter: '\r\n',
        });

        const readDevice = RNBluetoothClassic.onDeviceRead(device.address, onReceiveData);

        const onDisconnect = RNBluetoothClassic.onDeviceDisconnected(
          (event: any) => {
            readDevice.remove();
            onDisconnect.remove();

            dispatch(removeDevice(event.address));

            ToastAndroid.show(`${event.name} se ha desconectado`, 3000);
          },
        );

        dispatch(setConnectedDevice(device));
        resolve();
      } catch (error) {
        if (error instanceof Error) {
          ToastAndroid.show(error.message, 3000);
          reject(error.message);
        }
      }
    });
  };
```

I try to add this line of code on the disconnect function but I have the same problem: 

```ts
const disconnectDevice = (device: BluetoothDevice): Promise<void> => {
    return new Promise(async (resolve, reject) => {
      try {
        RNBluetoothClassic.onDeviceRead(device.address, onReceiveData).remove(); // <----
        await device.disconnect();
        
        dispatch(removeDevice(device.address));
        resolve();
      } catch (error) {
        if (error instanceof Error) {
          ToastAndroid.show(error.message, 3000);
          reject(error.message);
        }
      }
    });
  };
```

We need the user to be able to connect multiple devices. If you know of a different method I can fix or if it's a bug in the library please let me know.

ValentinOsvaldo, 2023-05-24:
Okey, sorry if my explanition is not clear...

I have one device, if I connect to the app, and later disconnect, but I connect the same device before de onDeviceDisconnect read the event, i have the same event when i received data duplicated, if I repeat the process, I have three messages, but the same device, only I disconnect and connect.

Is that my question or my problem. I see other issue and I copy the code to connect to the device where the ```onDataReceived``` listener and the ```onDeviceDisconnect```  listener is applied when connect the device.

If you want a video to show the "problem", I don't have problem to send you a video.
If is not a bug, what is the reason?, I understand that listeners are independent but they don't clean when I disconnect the device and the events affect the app behavior. (Sorry but english is not my native language).

ValentinOsvaldo, 2023-05-24:
Answering your question:
If connect a device and later disconnect the same device, but the ```onDeviceDisconnected``` read the event detect that device is disconnected, and later connect again the device, we have a normal behavior.
The problem is when the ```onDeviceDisconnected``` don't read the disconnected device (it takes a few seconds to detect it) and we try to connect the same device, what causes we have duplicated listeners.

From what I understood from your answer is that natively, we will have to prevent the same device from having two listeners at the same time. Am I correct?
I have no problem modifying what is necessary, only if you could tell me in which file or module I can make the change, I can do it without any problem.

kenjdavidson, 2023-05-25:
`The problem is when the onDeviceDisconnected don't read the disconnected device (it takes a few seconds to detect it) and we try to connect the same device, what causes we have duplicated listeners.`

This is due to the disconnection event being based on the `ACL Event` which from what I've read isn't the most accurate, but it's documented as such, so I don't think there is an issue with using it as long as this is the understanding.   With that said, I guess the issue is that:

- the device is disconnected 
- the connection is removed from `mConnections`
- the device is reconnected? 
- the device is disconnected?

I don't understand how this is possible?  I guess I'm still completely missing how you're getting two connections at the same time.  There is a check in there where:

- If attempting to connect to a device, and it's already in `mConnections` it will not allow you to connect again.

So you should never be able to have two active connections to the same device.  Now, there are some devices that will allow this:

- connect to device
- connect to device again (this causes an automatic disconnect)
- disconnect would fire
- connect would be successful

For example, this is how my old companies devices would work.  In this case, I needed to actually maintain the connection state globally in the react app, not just on the screen.

I don't see any way (easily, without breaking changes) to fix this on the Native side of things.

ValentinOsvaldo, 2023-05-31:
After a few days, I've some conclusions:

1. The problem can be solved saving a global state like redux where save the ```onDataReceived``` listener, example:

```ts
  ...
  listeners: {
    [address: string]: BluetoothEventSubscription;
  };
```

```ts
const readDevice = device.onDataReceived(onReceiveData);

dispatch(
  addListener({
    address: device.address,
    listener: readDevice,
  }),
);
```
On disconnect function

```ts
await device.disconnect();
listeners[device.address].remove();
```

**Note**: It's very important that when you connect to the device, than the connect button when is connecting you put the button disabled to avoid problems.

2. Another way is if you save the entry of the device, prevent two events that the timestamp and the address are the same. But if you don't do the first point, you can have a problem with the performance of your app.

3. If you only do the first point, you can have the same problem, but this only happened to me 1 time, and I things it is related to this

> **Note**: It's very important that when you connect to the device, than the connect button when is connecting you put the button disabled to avoid problems.

It should be noted that I have solved it, but it happened to me once, I don't know if it's the fault of the device, the application or performance itself.

I close the issue 'cause is not a bug but if anyone have the same problem keep this in mind.

Thanks @kenjdavidson for you help, I apologize because they are maybe small problems but I am a newbie

closed 2023-05-31 by ValentinOsvaldo

kenjdavidson, 2023-06-01:
Right on! Feel free to post about any issues or questions you have.  At this point, if more people think it should change, we can definitely change it.  But your definition is how I would expect this to be solved, without attempting to make assumptions about how the app will be used.   The less assumptions and more wide open the library can stay, I think is the best.