kenjdavidson/react-native-bluetooth-classic #258: Read from device as hex on Expo

Contresito, 2023-06-06:
Hey! First of all, thanks for this awesome library, it really is useful. 😊
Sorry if this has already been answered, couldn't find it anywhere!
I'm trying to read from a device and get back an array of hex values. I see that by default I get back an ascii string. That mostly works fine, because I can just translate it to hex again by using each characters' char code.
But, for the values that can't be translated into Unicode I get "FFFD" (�) which makes sense. That's why I would need to receive the values directly.
I read issue #50 but to do that I would need to eject from Expo...
Is there any other way or will I have to just bite the bullet?

kenjdavidson, 2023-06-06:
hahaha.. bite the bullet my friend.   Not possible to use the replacement configuration while in Expo to my knowledge.

Contresito, 2023-06-07:
Thanks for the quick reply!
I tried 
```js 
await newDevice.connect({ connectionType: 'binary' });
```
which I guess tells the library to use `ByteArrayDeviceConnectionImpl` instead of the default, and it actually changes what I receive from the device!
However, I still receive a string, and this time I'm struggling to see the correlation between what I got back and the bytes I should be getting...
If it helps, I'm getting back this string: `ASRaBAEWUVIHBiMBAQIAeo1GAAAAAP8AABkAAWgOAAAAAAAAFE4DDQo=`
that's 55 characters, but I'm expecting 41 bytes here...
`["1", "24", "4c", "4", "1", "17", "15", "53", "7", "6", "23", "1", "1", "2", "0", "7b", "fffd", "6", "0", "0", "0", "0", "fffd", "0", "0", "19", "0", "1", "68", "e", "0", "0", "0", "0", "0", "0", "14", "7e", "3", "d"]`
(that string and this array of values should be roughly equivalent, but some values are probably different)
How would I go about translating that string into the actual bytes I need? Maybe something obvious is flying over my head? 🤔

kenjdavidson, 2023-06-07:
Without looking at the code, I'm assuming that it's a base64 encoded string that you'll need to decode, then do you translations.   Maybe take a look at the PR and some of the other discussions about how this is managed.  There are a couple discussions with examples of how it's been used.

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/f47d8ff1d979321606f7474ec472319ba6d244c4/android/src/main/java/kjd/reactnative/bluetooth/conn/ByteArrayDeviceConnectionImpl.java#LL89C7-L89C69

Contresito, 2023-06-07:
Oh, base64, that was it! When I decoded it that way I got the correct string that I could then translate to hex digits. I didn't realize it was encoded!
Thanks a ton for your help Ken! You were a lifesaver.
I'm closing this now. :)

closed 2023-06-07 by Contresito

kenjdavidson, 2023-06-07:
Right on bud. You're welcome.

Closed #258<https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/258> as completed.