kenjdavidson/react-native-bluetooth-classic #259: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource

rahulp9538, 2023-06-12:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Any Android device
- OS: Android 12 or above

**Application Environment**
Provide information about your development environment:
- React Native version:  v0.65.1
- RN Bluetooth Classic version: 1.60.0-rc.28

**Describe the bug**
Your sample app doesnot work in Android 12 or above and gives below mentioned bug

Exception in native call
    java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSourc
[image]
e@2b036070: AdapterService getBondedDevices
        at com.android.bluetooth.Utils.checkPermissionForDataDelivery(Utils.java:482)
        at com.android.bluetooth.Utils.checkConnectPermissionForDataDelivery(Utils.java:514)
        at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.getBondedDevices(AdapterService.java:2122)
        at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.getBondedDevices(AdapterService.java:2114)
        at android.bluetooth.IBluetooth$Stub.onTransact(IBluetooth.java:642)
        at android.os.Binder.execTransactInternal(Binder.java:1285)
        at android.os.Binder.execTransact(Binder.java:1244)

kenjdavidson, 2023-06-12:
Have you attempted to add the correct permissions to it?

I have not updated the sample app with Android 12 permission updates, so it's not surprising it's missing something.  Feel free to add them an open a PR.  Otherwise, I'll try to get to it at some point (next time I need to start the app to do some development).

Kunal778, 2023-06-13:
I am facing same issue..

kenjdavidson, 2023-06-13:
Same thing, feel free to update the Permissions in the app for Android 12 (hopefully in a specific manifest) and submit a PR.  It's been a while, and I'm not actively working with React/Native at this point.   But I'd gladly accept the PR.

Kunal778, 2023-06-19:
Thanks, Buddy.!

Kunal778, 2023-06-19:
We have found the issue of this problem. This may help you. In the latest Android update, a new app permission is introduced called "Nearby Devices". Please open the app permissions and allow it to run successfully.

kenjdavidson, 2023-06-19:
There is a section in the Getting Started documentation https://kenjdavidson.com/react-native-bluetooth-classic/android/#android-permissions that shows how to request application permissions prior to using them.  Are you stating that:

- the `nearby devices` permission must be requested this way?
- OR that the `nearby devices` permission should be added to the manifest?
- OR both?

Kunal778, 2023-06-21:
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.respicoz">
     

     <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-feature android:name="android.hardware.bluetooth" android:required="true"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="false"/> 
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
     <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE"/>
       <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
  <uses-permission android:name="android.permission.FOREGROUND_SERVICE_LOCATION" />
  <uses-permission android:name="com.google.android.gms.permission.AD_ID"/>

    <application
      android:screenOrientation="portrait"
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:exported="true"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN"/>
            <category android:name="android.intent.category.LAUNCHER" />
            
        </intent-filter>
      </activity>
    </application>
</manifest>

Kunal778, 2023-06-21:
android {
    ndkVersion rootProject.ext.ndkVersion

    compileSdkVersion rootProject.ext.compileSdkVersion
    // compileSdkVersion 30 // You can use

    defaultConfig {
        applicationId 'com.respicoz'
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdk targetSdkVersion
        // targetSdkVersion 31
        versionCode 8
        versionName '2.1.0'
        multiDexEnabled true // <-- ADD THIS

Kunal778, 2023-06-21:
export const requestBluetoothEnabledFun = async () => {
  try {
    const isRequestBluetoothEnabled =
      await RNBluetoothClassic.requestBluetoothEnabled();
  
    return isRequestBluetoothEnabled;
  } catch (err) {
    // Handle accordingly
    
    return false;
  }
};

this function is not executed in Android 13 and higher versions. 
It seems like it needs the nearby devices permission.

kenjdavidson, 2023-06-21:
Thanks for this, I'll update the:
- Documentation
- Sample app

Next time I have a chance.  I'll leave this open so that I don't forget and others can find it easily.

vipulav-j, 2023-09-25:
This issue has been solved for me by putting runtime permission of BLUETOOTH_CONNECT and BLUETOOTH_SCAN as below.

```
if (Platform.OS === 'android' && Platform.Version >= 23) {
        const granted = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN && PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
          {
            title: 'permission required',
            message: 'permision is required to communicate',
            buttonPositive: 'ok',
          },
        );
        if (granted === PermissionsAndroid.RESULTS.GRANTED) {
          console.log('success scan');
        } else {
          console.log('scan not success');
        }
      }

Kunal778, 2023-10-02:
It is working fine now... Thank You :)..!!!

vipulav-j, 2023-10-04:
> It is working fine now... Thank You :)..!!!

are you able to send and receive messages with this library? are you doing for android only or for iOS as well? 
your reply will be very helpful to solve my issues.

kenjdavidson, 2023-10-04:
> > It is working fine now... Thank You :)..!!!
> 
> are you able to send and receive messages with this library? are you doing for android only or for iOS as well? your reply will be very helpful to solve my issues.

can you be more specific in what you're seeing wrong?  From your last comment you said that it as working.  IOS is a whole other beast, with a lot more hoops to jump through.  So if you're asking about IOS specifically, I suggest you take a look at the documentation and the MFi regulations for Apple bluetooth.

GabJank, 2024-01-12:
> This issue has been solved for me by putting runtime permission of BLUETOOTH_CONNECT and BLUETOOTH_SCAN as below.
>
> ```
> if (Platform.OS === 'android' && Platform.Version >= 23) {
>         const [...]

It works fine, but I really can't pair services anymore, if I pair in the configs of the phone I can do easily, but I can't pair anymore like I was in dispositives who had android < 12

sunny812546, 2024-02-06:
> > It is working fine now... Thank You :)..!!!
> 
> are you able to send and receive messages with this library? are you doing for android only or for iOS as well? your reply will be very helpful to solve my issues.

Yes. we do it only for Android. contact me if you need any help sunny84602@gmail.com.