kenjdavidson/react-native-bluetooth-classic #260: Method Not Implemented

JayashreePrabakaran, 2023-06-24:
**Mobile Device Environment**
- OS: IOS

**Application Environment**
- React Native version: [0.71.8]
- RN Bluetooth Classic version: [1.60.0-rc.28]

**Describe the bug**
When I try to use startDiscovery, getBondedDevices, getConnectedDevices, etc method, method not implemented error shows.

What I am missing? What I have to do ?

JayashreePrabakaran, 2023-06-24:
<key>UISupportedExternalAccessoryProtocols</key>
	<array>
		<string>com.apple.m1</string>
	</array>

	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>Access bluetooth device over serial connection to document connection features</string>
	
	
I've added these in my plist file

kenjdavidson, 2023-06-24:
I would suggest:
- reading the documentation
- reading about MFi 

those features are not supported on IOS.
that protocol is not a real protocol.

closed 2023-06-24 by kenjdavidson

JayashreePrabakaran, 2023-06-24:
I'm using this to connect with ESP32.

Where can I get the MFI Protocal for that? 
or this is not supported in IOS?

kenjdavidson, 2023-06-24:
I don't believe the esp32 is supported.

JayashreePrabakaran, 2023-06-24:
Is there any alternative to do that or  do you have any idea on how to do that?

kenjdavidson, 2023-06-24:
It's it's not mfi certified there is no way to do it with Bluetooth classic.

JayashreePrabakaran, 2023-06-24:
Hii

1. Even If I have MFI, We can achieve that?
2. I can't connect able to connect to ESP32 with IOS using this bluetooth classic module?
3. Or I can able to connect to ESP32 using any other module?

can you clarify this

kenjdavidson, 2023-06-24:
1.  If you have the mfi protocol it will work.
2.  Esp32 is not mfi certified so you can't have the protocol. So you cannot use this library.
3. Not that I'm aware of.

I suggest googling around a little.