kenjdavidson/react-native-bluetooth-classic #273: Connected devices always show as not connected

RobertGary1, 2023-10-12:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device:Android
- OS: 13

**Application Environment**
Provide information about your development environment:
- React Native version: 0.66.1
- RN Bluetooth Classic version: 1.60.0-rc.29

**Describe the bug**
If I do 
RNBluetoothClassic.getBondedDevices()
I get back all the bonded devices.
If I then call device.isConnected() on the bonded devices they always come back false

If I do
RNBluetoothClassic.getConnectedDevices()
it always comes back with 0 devices. Even though Android shows I have a couple devices connected

Could I have something wrong in my manifest that it cannot see connected devices?

    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>

kenjdavidson, 2023-10-12:
Suggest reading the docs:

Bonded = Bonded
Connected = open Serial connection

Connected does not mean what you think it means.

closed 2023-10-12 by kenjdavidson

RobertGary1, 2023-10-12:
Thank you. Is there anyway to determine if the phone is already connected to a device?

kenjdavidson, 2023-10-12:
That all depends what you mean by `connected`?  But the short answer is no.

Only one RFCOMM/serial connection can be opened to a device at one time, so if you have a serial connection open with a device from another application or other library, then there is no way to pass that off to this library.

If you're talking about `connected` in terms of being connected through one of the other profiles, then no, this library was specifically designed for RFCOMM communication, not any of the other profiles to which the android system would connect behind the scenes.