kenjdavidson/react-native-bluetooth-classic #274: Library crashes on project running on iOS 17 Iphone 15 pro

muhammaddadu, 2023-10-16:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: iPhone15 Pro
- OS: iOS 17

**Application Environment**
Provide information about your development environment:
- React Native version: RN 0.69.10
- RN Bluetooth Classic version: ^1.60.0-rc.29

**Describe the bug**
Application crashes when trying to run on device with this library installed on start.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

kenjdavidson, 2023-10-16:
Suggest reading the docs https://kenjdavidson.com/react-native-bluetooth-classic/ios/
and taking a look at the IOS MFi protocol requirements.

This error is expected since you, most likely, have not provided the appropriate protocol strings in your application.  You can look at the sample app to see how that works.

closed 2023-10-16 by kenjdavidson

muhammaddadu, 2023-10-16:
This is to do with the new strict Swift rules in the latest XCode.

This changed resolved it

```
self.supportedProtocols = (Bundle.main
            .object(forInfoDictionaryKey: "UISupportedExternalAccessoryProtocols") as? [String]) ?? [];
```

CC @kenjdavidson

muhammaddadu, 2023-10-16:
> Suggest reading the docs https://kenjdavidson.com/react-native-bluetooth-classic/ios/ and taking a look at the IOS MFi protocol requirements.
>
> This error is expected since you, most likely, have not [...]

Please re-open this ticket as the issue is something else.

kenjdavidson, 2023-10-16:
All this is doing is forcing to an empty list, I much prefer it to error out so that the developer knows something is missing.  With this change, it will compile and start fine, but there will be way more questions about why the devices aren't being found, etc.

Personally I'd prefer compile time errors rather than runtime issues.

muhammaddadu, 2023-10-16:
Thank you. Makes sense. Let me try this and get back to you :)

muhammaddadu, 2023-10-16:
Wonder if their is a way to trigger the react-native error screen to highlight you haven't enabled `UISupportedExternalAccessoryProtocols` 🤔

If I can get some time, will try and contribute to this project.