kenjdavidson/react-native-bluetooth-classic #275: Cannot read property 'getBondedDevices' of null

woflmao, 2023-10-17:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Iphone 12 mini
- OS: ios 16.6.1

**Application Environment**
Provide information about your development environment:
- React Native version: v0.72.5
- RN Bluetooth Classic version: v1.60.0-rc.29

**Describe the bug**
Android worked fine, when I switched Expo to build ios, I am getting "Cannot read property 'getBondedDevices' of null" when attempting to use RNBluetoothClassic.getBondedDevices(). Upon logging RNBluetoothClassic I get this result:

`{"_eventEmitter": {"_nativeModule": {}}, "_nativeModule": null}`

As opposed to the same call on Android which produces this:

`{"_eventEmitter": {"_nativeModule": {}}, "_nativeModule": {"accept": [Function promiseMethodWrapper], "addListener": [Function nonPromiseMethodWrapper], "availableFromDevice": [Function promiseMethodWrapper], "cancelAccept": [Function promiseMethodWrapper], "cancelDiscovery": [Function promiseMethodWrapper], "clearFromDevice": [Function promiseMethodWrapper], "connectToDevice": [Function promiseMethodWrapper], "disconnectFromDevice": [Function promiseMethodWrapper], "getBondedDevices": [Function promiseMethodWrapper], "getConnectedDevice": [Function promiseMethodWrapper], "getConnectedDevices": [Function promiseMethodWrapper], "getConstants": [Function anonymous], "isBluetoothAvailable": [Function promiseMethodWrapper], "isBluetoothEnabled": [Function promiseMethodWrapper], "isDeviceConnected": [Function promiseMethodWrapper], "openBluetoothSettings": [Function nonPromiseMethodWrapper], "pairDevice": [Function promiseMethodWrapper], "readFromDevice": [Function promiseMethodWrapper], "readMultipleFromDevice": [Function promiseMethodWrapper], "readOneFromDevice": [Function promiseMethodWrapper], "removeAllListeners": [Function nonPromiseMethodWrapper], "removeListener": [Function nonPromiseMethodWrapper], "requestBluetoothEnabled": [Function promiseMethodWrapper], "setBluetoothAdapterName": [Function promiseMethodWrapper], "startDiscovery": [Function promiseMethodWrapper], "unpairDevice": [Function promiseMethodWrapper], "writeToDevice": [Function promiseMethodWrapper]}}`

**To Reproduce**
Steps to reproduce the behavior:
1. Call RNBluetooth.getBondedDevices() on an ios device

**Expected behavior**
Expect to return a list of bonded devices

kenjdavidson, 2023-10-17:
From my understanding there are a bunch of issues related to Bluetooth and Expo.   There was a secondary/addon library documented in order to get it working https://kenjdavidson.com/react-native-bluetooth-classic/expo/ you'll need to look into that.   I don't do much Mobile development anymore, and I've never used Expo, so this is out of my realm.

Also, have you followed all the required functionality for IOS, ie. MFi protocol strings?  This may be compiling on Expo but failing here instead of at the protocol setting portion.

woflmao, 2023-10-17:
Yes I have the MFi protocol string in the expo plugins "with-rn-bluetooth-classic" section, thanks for the quick response though I will do some more research!

kenjdavidson, 2023-10-17:
Can you email me the protocols?
What's the device?

The error implies that the native module isn't loading. Have you tired ejecting and doing locally?

kenjdavidson, 2023-10-18:
Following up for those that may come across this issue.  It looks like the IOS side of things is not auto-linking the native module.   @woflmao is going to attempt to:
- Reinstall the POD and hope that it picks up
- Attempt to manually link using the appropriate React Native docs

Chances are that React Native changed how Auto linking of native modules works, I recently read something in 0.70 version about native modules being changed.  If this is the case, I'll attempt to break off:

- `0.64.x-rcY` to maintain lower version
- `0.70.x-rcY` to start looking at fixing how React native manages native modules.

PRs will be appreciated if so.

woflmao, 2023-10-18:
Update: The steps shared by @kenjdavidson above did work to solve this issue. Thanks again for all the help.

closed 2023-10-18 by kenjdavidson

ShaneZhengNZ, 2024-05-13:
@woflmao I am having the same issue. I am using Expo managed workflow with dev client. Do you have an idea about what I need to do to fix the issue?