kenjdavidson/react-native-bluetooth-classic #276: Does not find following device

HarunKilic, 2023-10-17:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: [Samsung S23]
- OS: [SDK 33]

**Application Environment**
Provide information about your development environment:
- Expo SDK 49
- RN Bluetooth Classic version: [1.60.0-rc.29]

**Describe the bug**
Product: [Eyoyo EY-016](https://www.eyoyousa.com/eyoyo-ey-016-2d-finger-ring-barcode-scanner-mini-wearable-3-in-1-usb-wired-amp-24g-wireless-amp-bluetooth-scanner-image-1d-qr-bar-code-reader-pdf417-data-matrix-screen-scan-for-ipad-smartphone-pc-p00212p1.html)
I can find this device in native bluetooth search on samsung device.
But I can't find it with this library, it has bluetooth 4.1 compability.

**To Reproduce**
Steps to reproduce the behavior:
1. Get the device
2. Try to scan devices
3. Do not see this device. If you see it, close your eyes :P

kenjdavidson, 2023-10-17:
Have you followed the expo docs?
Have you ejected to see if that works?
Do you get any errors?
Have you debugged it?

HarunKilic, 2023-10-17:
> Have you followed the expo docs?
> Have you ejected to see if that works?
> Do you get any errors?
> Have you debugged it? 

Hi, yes I did. Did see some devices but not the one I was looking for. 

After some time, it did actually show up! So it is working now as expected! This can be closed now.

kenjdavidson, 2023-10-18:
Right on, glad it worked out my man.

closed 2023-10-18 by kenjdavidson