kenjdavidson/react-native-bluetooth-classic #286: Method does not override or implement a method from a supertype @Override

mMarcos208, 2023-12-21:
**Mobile Device Environment**
Android 14

**Application Environment**
"react-native-bluetooth-classic": "^1.60.0-rc.30",
"react-native": "0.73.1",
"react": "18.2.0",

**Describe the bug**
node_modules/react-native-bluetooth-classic/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java:233: error: method does not override or implement a method from a supertype @Override 

**To Reproduce**
Steps to reproduce the behavior:
1. yarn android

kenjdavidson, 2023-12-21:
I assume you clicked enter too quickly, can you please clean up the issue and provide more/complete details on what the problem is?

kenjdavidson, 2023-12-21:
You're also using `1.60` with `react-native:0.73`, chances are they changed the `NativeModule` at some point around `0.70` which means:
- You'll need to revert back to an older version of react native
- I'll have to actually spend some time in upgrading the library to be compatible with `0.7x` 

I'd be willing to accept a PR updating the library to be compatible with `0.7x` of react native, but I don't have the time to do so at the moment.

mMarcos208, 2023-12-21:
@kenjdavidson react native 0.72.7 works well. Maybe upgrade dependences works.

kenjdavidson, 2023-12-21:
That means they changed the Java classes or interfaces in.0.73. Which me as the library isn't supported until it gets updated.  I'll have to take a look and see how much work that will be.   I read an article somewhere they were completely changing how native modules worked.

laooola, 2023-12-23:
having the same problem, blocks me from upgrading to 0.73. happy to assist if you need further information.

kenjdavidson, 2023-12-23:
Feel free to open a pr updating the library to be compatible with 0.73.ill gladly review and release.

kenjdavidson, 2024-01-10:
I started working on the update to 0.73.  I started creating the following branches:

### [https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/tree/react-native-073](https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/tree/react-native-073)

Can use this to do the development.  It looks like react 0.73 has better support for Linked modules, so no more effing around with Metro custom configuration, should just be able to `npm link` the project.

### [https://github.com/kenjdavidson/react-native-bluetooth-classic/tree/react-native-073](https://github.com/kenjdavidson/react-native-bluetooth-classic/tree/react-native-073)

Available for the development.   It looks like there are two things:

1. need to get all the Module classes update to appropriately implement all the right things.  Once this is done, HOPEFULLY the current auto linking just works.

2. It looks like they've completely re-engineered the way native modules are loaded.  Need to look into this, but it could mean that it won't be possible to manually load the module and therefore won't be able to add custom Connection types.  Which would be sad, but I'm sure there's still a way to extend the library.

kenjdavidson, 2024-01-10:
> having the same problem, blocks me from upgrading to 0.73. happy to assist if you need further information.

Feel free to shoot me a message:
- Email
- Skype

Whatever works for you if you're interested in working with it.  It would be much appreciated.

kenjdavidson, 2024-01-23:
Should be resolved, hopefully.  If there are any issues with 0.73 please open another task with specifics.

closed 2024-01-23 by kenjdavidson