kenjdavidson/react-native-bluetooth-classic #287: android.permission.BLUETOOTH_CONNECT permission for AttributionsSource

KartikDevarde2601, 2023-12-31:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: [e.g. Samsung A13]
- OS: [Android 14 ]

**Application Environment**
Provide information about your development environment:
- React Native version: [ v0.71.0]
- RN Bluetooth Classic version: [. 1.60.0-rc.30]

**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
when we call  

const PairDevice = async () => {
      const devices = await RNBluetoothClassic.getBondedDevices();
      console.log(devices);
    };
get this `error`

**Screenshots**
[image]

**Additional context**
import React from 'react';
import {useState, useEffect} from 'react';
import {View, Text, SafeAreaView, ToastAndroid} from 'react-native';
import RNBluetoothClassic, {
  BluetoothDevice,
} from 'react-native-bluetooth-classic';
import {PermissionsAndroid} from 'react-native';

const requestBluetoothPermissions = async () => {
  const fineLocationGranted = await PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
    {
      title: 'Access fine location required for discovery',
      message:
        'In order to perform discovery, you must enable/allow ' +
        'fine location access.',
      buttonNeutral: 'Ask Me Later"',
      buttonNegative: 'Cancel',
      buttonPositive: 'OK',
    },
  );

  return fineLocationGranted === PermissionsAndroid.RESULTS.GRANTED;
};

const App = () => {
  const [isEnabled, setIsEnabled] = useState(false);

  useEffect(() => {
    async function checkBluetoothEnabled() {
      const granted = await requestBluetoothPermissions();
      console.log(`permissions granted: ${granted}`);
      const enabled = await RNBluetoothClassic.isBluetoothEnabled();
      setIsEnabled(enabled);
      console.log(`phone Bluetooth is ON/OFF: ${enabled}`);

      if (!enabled && granted) {
        ToastAndroid.show(
          'Bluetooth is OFF, please turn it ON',
          ToastAndroid.SHORT,
        );
      }
    }

    const PairDevice = async () => {
      const devices = await RNBluetoothClassic.getBondedDevices();
      console.log(devices);
    };

    checkBluetoothEnabled();
    PairDevice();
  }, []);

  return (
    <SafeAreaView>
      <View>
        <Text>App</Text>
      </View>
    </SafeAreaView>
  );
};

export default App;

kenjdavidson, 2023-12-31:
Have you tried adding the permission to your app?  Any changes to your manifest permission will be added to the libraries and it should work.

KartikDevarde2601, 2023-12-31:
Hi, thanks for helping me solve the error. I was using Android 14 and missing the targetSdkVersion in the manifest, causing an issue for Android 12+
 `<manifest>
<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
</manifest>`

closed 2023-12-31 by KartikDevarde2601