kenjdavidson/react-native-bluetooth-classic #290: Intermittent in onDataReceived listener ESP32 - HC05

junioroga, 2024-01-08:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android Redmi Note 9S
- OS: MIUI 12 SKQ1.211019.001

**Application Environment**
Provide information about your development environment:
- React Native version: 0.72.6
- RN Bluetooth Classic version: 1.60.0-rc.30

**Describe the bug**
I have an ESP32 connected to an HC 05 that returns data from a temperature sensor when I send the string "L". I have a testing application developed on native Android, which works perfectly.
I'm migrating to React Native and I'm using this lib.
When connecting for the first time, the onDataReceived listener does not return any value. If I disconnect and connect again, it works.
If I keep connecting and disconnecting several times, I notice that onDataReceived often does not return.
In the native application, I can do this as many times as I want, and the feedback comes back perfectly, removing any doubts about the problem being in the ESP32.

Connect function
[image]

Saving conection and sending the string
[image]

Writing on bluetooth
[image]

onDeviceRead function
[image]

I've looked at several examples, I've tested different ways to get around the situation and I always have the same behavior.
I've already tried putting read() in a loop with a 100ms interval instead of using the listener, but the same thing happens, sometimes it works, sometimes it doesn't.
Maybe someone can help me?

kenjdavidson, 2024-01-08:
Bluetooth code is pretty specific, I've never seen this type of behavior while working in the current testing app, nor the application in which I designed the library for.   I'm not entirely sure what you mean by:

```
If I keep connecting and disconnecting several times, I notice that onDataReceived often does not return.
```

Do you mean that:

* The Promise from calling `await subscription = device.onReadData()` doesn't return?
* Once the subscription is created the listener is never called?

These two things are very separate.  In both cases, this will require you debugging the Native code to see what is happening.  The location you debug is based on the answer to the above question.

Feel free to update the task with any debugging details:

* Line numbers
* Data for variables at line numbers
* pretty much standard debugging things

And if you find an issue feel free to open a PR.  Otherwise I'll do my best to assist your debugging efforts.

junioroga, 2024-01-08:
The first connection, the onDataReceived listener does not return. Remaining undefined as it is started.

[image]

When I disconnect and connect again, then it works.

[image]

What would this onDataRead method be that you mentioned? What I have here only has onDataReceived.

kenjdavidson, 2024-01-08:
Ya, that method, I don't remember the exact names, but you get the point.

Not sure why it would return undefined sometimes, unless something is going on with the promises, etc.  Have you actually debugged it to see what's going on?  Like debugged into the `onDataReceived` method?

```
  /**
   * Adds a listener to the device.  Once completed this will:
   * - send queued data already read from the device (if implemented by DeviceConnection)
   * - send all subsequent data
   *
   * @param listener the BluetoothEventListener which will receive incoming data
   */
  onDataReceived(
    listener: BluetoothEventListener<BluetoothDeviceReadEvent>
  ): BluetoothEventSubscription {
    return this._bluetoothModule.onDeviceRead(this.address, listener);
  }
```

It calls through to the bluetooth module, which you can also debug.

junioroga, 2024-01-08:
Would I do this debug using vscode? I have never used breakpoints to debug in RN. Can we debug even the native part code?
Can you help me understand better?

kenjdavidson, 2024-01-08:
You can debug the Javascript portion in VSCode, and you can debug the Native portion in Android Studio.

kenjdavidson, 2024-01-10:
Any word on this?

junioroga, 2024-01-10:
Hello, unfortunately I won't be able to debug. I was developing the application for an MVP of a product, and the owner decided to start using it even though the connection was intermittent. So he took the Arduino and now I have no way to test it.
He brought me a new one, and as soon as it arrives, I will carry out the steps you said.
For now I think we can close the issue.

closed 2024-01-10 by kenjdavidson