kenjdavidson/react-native-bluetooth-classic #291: Can't pair but can connect.

GabJank, 2024-01-12:
**Mobile Device Environment**
  - Device: [e.g. Android, Android]
  - OS: [e.g. API30, Red Velvet, API31, Snow Cone]

**Application Environment**
  "react": "18.2.0",
  "react-native": "0.73.1",
  "react-native-bluetooth-classic": "^1.60.0-rc.30",

**Describe the bug**
  So I was working in a android 11 phone and it was working perfectly, but I was going to test in a android 12 and it can't connect. I was using the example to test somethings with the bluetooth but can't really understand what is wrong.

There are my permissions:

```
const requestAccessFineLocationPermission = async () => {
  const bluetoothScanPermission = await PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN
  );
  const bluetoothConnectPermission = await PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
  );
  const fineLocationPermission = await PermissionsAndroid.request(
    PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
  );
  return (
    fineLocationPermission === "granted" &&
    bluetoothScanPermission === "granted" &&
    bluetoothConnectPermission === "granted"
  );
};
```

And it gaves the error: "Connection failed: java.io.IOException: read failed, socket might closed or timeout, read ret: -1

I just really don't understand why it doesn't work because in a android 11 worked perfectly and in the permissions of the app it has the option of pair to devices active.

kenjdavidson, 2024-01-12:
Read failed usually means the other side of the socket isn't open for connection.  But as with all things, I have limited devices and time a d access to your code. So you'll have to debug for yourself, I'm more than willing to help or give ideas, but these types of issues need more work on your part.

kenjdavidson, 2024-01-12:
Also, 0.73 isn't supported (I don't thing) I don't even know how you're starting the app. There's a other task states there is an issue with the new native interfaces.

I just don't have time to do the upgrade myself a si no longer work on the app I built this for.

closed 2024-01-23 by kenjdavidson