kenjdavidson/react-native-bluetooth-classic #292: Unable to install with Expo 5- (rn 0.73) Deprecated gradle features were used.

mrlmark, 2024-01-21:
**Application Environment**
- React Native version: 0.73.0
- RN Bluetooth Classic version: react-native-bluetooth-classic@^1.60.0-rc.30

**Describe the bug**
[RUN_GRADLEW] FAILURE: Build failed with an exception.
[RUN_GRADLEW] * What went wrong:
[RUN_GRADLEW] Execution failed for task ':react-native-bluetooth-classic:compileDebugJavaWithJavac'.
[RUN_GRADLEW] > Compilation failed; see the compiler error output for details.
[RUN_GRADLEW] * Try:
[RUN_GRADLEW] > Run with --info option to get more log output.
[RUN_GRADLEW] > Run with --scan to get full insights.
[RUN_GRADLEW] BUILD FAILED in 37s
[RUN_GRADLEW] Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

kenjdavidson, 2024-01-21:
Yup. 0.73 is not supported. I have a task to update but no time line. Feel free to help out with the process.

closed 2024-01-21 by kenjdavidson

serfgy, 2024-02-03:
was this fixed in 0.73 rc-4? i used that with expo 50 and it still says gradle build error