kenjdavidson/react-native-bluetooth-classic #298: Unable to resolve module ../../react-native/types/index

maartenvandillen, 2024-01-25:
**Mobile Device Environment**
All, it is a javascript error

**Application Environment**
Provide information about your development environment:
- React Native version: 0.72.6
- RN Bluetooth Classic version: 1.73.0-rc.3

**Describe the bug**
ModuleResolver cannot find javascript import in file lib/index.js:
import { NativeModules } from "../../react-native/types/index";

and in lib/BluetoothModule.js
import { NativeEventEmitter, Platform } from "../node_modules/react-native/types/index";

**To Reproduce**
Steps to reproduce the behavior:
1. Add package to project
2. Run project

**Expected behavior**
Project builds

**Additional context**
Fixed by changing imports to:
import { NativeModules } from "react-native";
and
import { NativeEventEmitter, Platform } from "react-native";

kenjdavidson, 2024-01-25:
Rnvc 73 is not compatible with react 72. Use the older 0.60 version. See the matrix on the Readme.

closed 2024-01-25 by kenjdavidson

kenjdavidson, 2024-01-25:
Or I messed things up when fixing the typescript errors to get it to compile.   Either way I'd give the older version a try with react native 72.

maartenvandillen, 2024-01-25:
Ok I will try that. Will I be missing out on bug fixes or new features? If so, I can live with the patched version I have right now.

kenjdavidson, 2024-01-25:
No.

But if you've fixed it locally. Open a pr.

kenjdavidson, 2024-01-25:
I think I fixed the typing issue with react-native that I introduced when attempting to both merge the update and resolve outdated libraries.  Anyhow, no more TS issues and it looks like the import statements are correct now.