kenjdavidson/react-native-bluetooth-classic #302: While trying to resolve module `react-native-bluetooth-classic` was successfully found. However, this package itself specifies a `main` module field that could not be resolved

mfventura, 2024-02-05:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android 13 (API 33)

**Application Environment**
Provide information about your development environment:
- React Native version: 0.73.3
- RN Bluetooth Classic version: 1.73.0-rc.4

**Describe the bug**
When I try to import the components from 'react-native-bluetooth-classic', it´s not possible to run the app.
I get a trace that says:
`error: Error: While trying to resolve module `react-native-bluetooth-classic` from file `/Users/theam/Cabritas/src/TestBT.tsx`, the package `/Users/theam/Cabritas/node_modules/react-native-bluetooth-classic/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`/Users/theam/Cabritas/node_modules/react-native-bluetooth-classic/lib/index.js`. Indeed, none of these files exist:

  * /Users/theam/Cabritas/node_modules/react-native-bluetooth-classic/lib/index.js(.android.jsx|.native.jsx|.jsx|.android.js|.native.js|.js|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx|.android.cjs|.native.cjs|.cjs|.android.json|.native.json|.json)
  * /Users/theam/Cabritas/node_modules/react-native-bluetooth-classic/lib/index.js/index(.android.jsx|.native.jsx|.jsx|.android.js|.native.js|.js|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx|.android.cjs|.native.cjs|.cjs|.android.json|.native.json|.json)
    at DependencyGraph.resolveDependency (/Users/theam/Cabritas/node_modules/metro/src/node-haste/DependencyGraph.js:283:17)
    at /Users/theam/Cabritas/node_modules/metro/src/lib/transformHelpers.js:176:21
    at resolveDependencies (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:56:25)
    at visit (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:107:30)
    at async Promise.all (index 2)
    at async visit (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:116:5)
    at async Promise.all (index 0)
    at async buildSubgraph (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:127:3)
    at async Graph._buildDelta (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/Graph.js:249:22)
    at async Graph.initialTraverseDependencies (/Users/theam/Cabritas/node_modules/metro/src/DeltaBundler/Graph.js:227:19)`

**To Reproduce**
1. Create a fresh project, using react-native 0.73, Android API 33 and JDK 17. https://reactnative.dev/docs/environment-setup
2. Run the project successfully.
3. In the project, create a folder 'src'
4. Create a TestBT.tsx file
5. Modify index.js to import App from src/TestBT instead of App
6. Run the project

**Expected behavior**
I expected to see the TestBT.tsx text without errors.

apcrtc, 2024-02-05:
having the same exact issue, somehow it seems like "lib" folder in version 1.73.0-rc.4 does not exists. check package.json > "main": "lib/index.js"

mfventura, 2024-02-05:
> having the same exact issue, somehow it seems like "lib" folder in version 1.73.0-rc.4 does not exists. check package.json > "main": "lib/index.js"

Yes, I saw it, but I don's really know how to fix it (I´m more related with Java in backend projects).

Were you able to fix that? Maybe using a minor version?

apcrtc, 2024-02-05:
Well... I rolled back to version to rc3, the lib folder exists in this RC and no initial errors but upon running metro I see that TS types are messed up. Its not good..

Try: npm install react-native-bluetooth-classic@1.73.0-rc.3

mfventura, 2024-02-05:
Yes, trying that, I get an error on runtime.

´´´
error: Error: Unable to resolve module ../node_modules/react-native/types/index from /Users/theam/personal-projects/cabritas/node_modules/react-native-bluetooth-classic/lib/index.js: 

None of these files exist:
  * node_modules/react-native-bluetooth-classic/node_modules/react-native/types/index(.android.jsx|.native.jsx|.jsx|.android.js|.native.js|.js|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx|.android.cjs|.native.cjs|.cjs|.android.json|.native.json|.json)
  * node_modules/react-native-bluetooth-classic/node_modules/react-native/types/index/index(.android.jsx|.native.jsx|.jsx|.android.js|.native.js|.js|.android.ts|.native.ts|.ts|.android.tsx|.native.tsx|.tsx|.android.cjs|.native.cjs|.cjs|.android.json|.native.json|.json)
> 1 | import { NativeModules } from "../node_modules/react-native/types/index";
    |                                ^
  2 | import BluetoothDevice from "./BluetoothDevice";
  3 | import BluetoothError from "./BluetoothError";
  4 | import { BluetoothEventType, } from "./BluetoothEvent";
    at ModuleResolver.resolveDependency (/Users/theam/personal-projects/cabritas/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:127:15)
    at DependencyGraph.resolveDependency (/Users/theam/personal-projects/cabritas/node_modules/metro/src/node-haste/DependencyGraph.js:271:43)
    at /Users/theam/personal-projects/cabritas/node_modules/metro/src/lib/transformHelpers.js:176:21
    at resolveDependencies (/Users/theam/personal-projects/cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:56:25)
    at visit (/Users/theam/personal-projects/cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:107:30)
    at async Promise.all (index 9)
    at async visit (/Users/theam/personal-projects/cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:116:5)
    at async Promise.all (index 2)
    at async visit (/Users/theam/personal-projects/cabritas/node_modules/metro/src/DeltaBundler/buildSubgraph.js:116:5)
    at async Promise.all (index 0)
´´´

kenjdavidson, 2024-02-05:
Rc4 is using the typescript parent from react native. Which probably has a different dist folder configured and why the lib folder is missing.  So obviously fucked up while attempting to fox old dependencies. 

I'm on vacation and can accept a pr, but can't do much else.  If you're looking into it, check the typescript folder as this is probably where the lib issue is.

kenjdavidson, 2024-02-05:
Rc3 has imports pointing to the types and not the actual code, vscode "fixed" this when I first removed the @types/reactnative in favor of the built in types.

kenjdavidson, 2024-02-05:
Looking at the tsconfig it's still got /lib as the outdir.  So that seems fine. It looks like /lib is in the git ignore, which I can't remember if this causes issues for npm as well.

apcrtc, 2024-02-05:
I have cloned the package and ran tsc but there is no output. It seems like tsc is not outputting lib folder. Looking at "noEmit" flags etc at the moment. By the way I noticed you extend tsconfig flags from "@tsconfig/react-native/tsconfig.json" in 0.73 this was replaced with: "@react-native/typescript-config/tsconfig.json" I suggest this is replaced. If I have any update I will let you know.

kenjdavidson, 2024-02-05:
Ah I tried to update the npm ignore to force the lib dir.  When I tested locally last it compiled OK. Open a pr of you get it built.

apcrtc, 2024-02-05:
Opened a PR to fix file emitting please review: https://github.com/kenjdavidson/react-native-bluetooth-classic/pull/303

I have tested locally with local npm package publish and works fine. Try building yourself: ```npx tsc --build --verbose```

kenjdavidson, 2024-02-05:
Merged and kicked off release. Appreciate the help. And sorry for the last few builds. My bad.

apcrtc, 2024-02-05:
Thanks for fast response.
I can confirm that tsc is outputting it properly now and I can use the package in 0.73^. 

When I pulled the package I had a different TS error this time, I have corrected the NativeModule import path, please review: https://github.com/kenjdavidson/react-native-bluetooth-classic/pull/304

With this fix all issues should be resolved for now.

kenjdavidson, 2024-02-05:
Sorry was pool side. Merged and released.

apcrtc, 2024-02-05:
Thank you. For me issue is completely resolved.

Thank you for maintaining library, its very useful and widely used in our company! Enjoy your holiday 😄

apcrtc, 2024-02-05:
It seems like import issue is still persisting. I have checked the lib/index.js but old import statement is still there.

[image]

After my PR it should have been like this:

[image]

Is it possible that you might have forgotten run tsc?

For example I created this package for myself to test issues, if you check lib/index.js you will find corrected import statement: https://www.npmjs.com/package/@alihanpehlivan/react-native-bluetooth-classic

kenjdavidson, 2024-02-05:
I would take a look at the github action. This is what builds and releases the library. If it's not correct with the latest changes it may be wrong.

apcrtc, 2024-02-05:
It seems like last time actions worked was 6:40pm UK time. My latest PR was at around 7:20pm. I think actions did not take the latest change. Not sure how did that happen.

kenjdavidson, 2024-02-05:
Kicked it off again.

apcrtc, 2024-02-05:
There we go, now that worked! Not sure why it did not kick off properly tsc when you merged into master, interesting.
I can see index.js updated now and ever

I can confirm that all issues related to imports & srcs are resolved now and I have tested it with my devices.
I'm glad I can put all our teams projects on 0.73 now.

Thanks for maintaining the library, enjoy your holiday! 😃

kenjdavidson, 2024-02-05:
No. Thank you. I don't do a lot with this anymore.But I'm glad it's used.

closed 2024-02-05 by kenjdavidson