kenjdavidson/react-native-bluetooth-classic #305: Doubt: Enable plugin only for Android?

EduFrazao, 2024-02-12:
Hi. Thanks for this very useful library!
At the moment, I only need to support bluetooth usage on Android, but my APP needs to work on IOS too.
Apple is preventing me to publish my application, because of the links of this library with MFi.

Its possible to unlink this library on IOS and use it only on Android?

kenjdavidson, 2024-02-12:
Maybe, if you remove all the autolinked dependencies in the IOS project and build it manually.   But there is nothing in the project that would do so, you'd have to find a way to overwrite or ignore the `podspec` so that it doesn't load.   I don't know if this will cause any problems downstream (in javascript) since for Android the Native Module will be there, but for IOS the native module will be null.  I guess as long as you're checking.

At this point I'm going to close the task, but feel free to come back and update it if you see a way to do so.  I would suggest if you cant find anything, give the original library (which this was forked from) `react-native-bluetooth-serial` as it uses bluetooth classic on Android and falls back to BLE on IOS.

closed 2024-02-12 by kenjdavidson

maartenvandillen, 2024-02-28:
Hi @EduFrazao same use case here. How did you solve this?

kenjdavidson, 2024-02-28:
I'll try to add this to the documentation at some point, thanks for looking up the autolinking functionality.