kenjdavidson/react-native-bluetooth-classic #306: Futur for expo ? (Install on SDK 49 ?)

seba9999, 2024-02-13:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device:  Android
- OS: API>33

**Application Environment**
Provide information about your development environment:
- React Native version: 0.72.6
- RN Bluetooth Classic version: v1.73.0-rc.7

**Describe the bug**

I'm trying to install this on my SDK49 expo app but It fails with an `unable to resolve dependency tree` because SDK49 is `react-native@"0.72.6"`  and the build is aiming for `react-native@">=0.73.1"`

I saw in this abandonned PR that you found that SDK50 won't be working anymore ...

https://github.com/kenjdavidson/react-native-bluetooth-classic/pull/293/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R19

Could it work with a previous release ? v1.60.0-rc.30 maybe ?

What are the futur for this lib for expo SDK50+ ?

kenjdavidson, 2024-02-13:
Let me preface this by ... I've never used Expo.  This library and the app it was written for was built outside of expo, so anything expo related is a black hole to me, and I rely on others that use it in order to guide the libraries usage.

That PR was rejected because the guys that update it:
- made changes based on an old branch, making it super complicated to merge
- updated things like package name that would have affected the NPM release

If you want to reach out to them to see if they'll redo the changes then feel free.  Or if you'd like to use their release of the same library, feel free.

Looking at your configuration

```
React Native version: 0.72.6
RN Bluetooth Classic version: v1.73.0-rc.7
```

these two versions are incompatible for sure.  I would suggest trying:

```
React Native version: 0.72.6
RN Bluetooth Classic version: v1.60.x
```

as documented in the table on the home page.

closed 2024-02-13 by kenjdavidson

seba9999, 2024-02-13:
I'm sorry man ! I've understand this table a little too late ! 
Forgive me and accept this tiny PR to help not getting more issue open like this dumb mine ! :D 

https://github.com/kenjdavidson/react-native-bluetooth-classic/pull/307

kenjdavidson, 2024-02-13:
ha.. right on, not a problem.